ETLCPP/etl #785: Add etl::duration?

jaskij, 2023-11-09:
Just today, I was greeted by a nasty message from GCC 13.2 that the `<chrono>` header (and thus everything in `std::chrono`) is not available in a freestanding implementation.

While that is true, it *has* worked in the past. Would it make sense to add an `etl::duration`? It's the part of it that previously worked, and the one most likely to be missed. An alternative option would be to encourage the use of a units library (either generally, or a specific one).

jwellbelove, 2023-11-10:
I started a feature branch a while ago (feature/#757-Add-time-date-classes) that I plan to gradually add to over time.

It currently just implements some of the basic chrono units:-

day
weekday
weekday_indexed
weekday_last
month
year
last_spec
duration

jaskij, 2023-11-10:
Turns out it was just an issue with compilation flags (don't use `-ffreestanding` with GCC and newlib), so this issue is moot.

closed 2023-11-10 by jaskij