ETLCPP/etl #786: u8string

wh1t3lord, 2023-11-11:
Hello and thanks for putting so much effort into this project!

By the way, it would be perfect to see a header file of u8string.  

Is any reason for not adding this? Maybe I don't know something, because I tried to find explanation related to this class and the result was nothing. For that case I want to ask about possible adding of u8string class. Because in C++20 we have the char8_t type and it's separated to char. 

I think it is a reasonable thing today, library has u16string, u32string and wstring, but u8string doesn't exist.

jwellbelove, 2023-11-11:
It should be pretty easy to add `etl::u8string`. I'll look into it.

jwellbelove, 2023-12-05:
Added 20.38.7

closed 2023-12-05 by jwellbelove