ETLCPP/etl #787: etl::expected doesn't compile with ETL_LOG_ERRORS

HMS-JJO, 2023-11-13:
Hi,

When enabling the `ETL_LOG_ERRORS` option I am unable to use etl::expected since `expected_invalid<TError>` doesn't publicly inherit from `expected_invalid<void>` resulting in the following error:

```
etl/error_handler.h:299:66: error: 'etl::exception' is an inaccessible base of 'etl::expected_invalid<Type>'
     #define ETL_ASSERT(b, e) {if(!(b)) {etl::error_handler::error((e));}}                                 // If the condition fails, calls the error handler
                                         ~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
etl/expected.h:695:7: note: in expansion of macro 'ETL_ASSERT'
       ETL_ASSERT(storage.index() == Value_Type, ETL_ERROR(expected_invalid<TError>));
```

I believe there is a missing `public` in `etl/expected.h:77`.

This affects etl version `20.38.6`.

jwellbelove, 2023-12-05:
Fixed 20.38.8

jwellbelove, 2023-12-05:
Fixed 20.38.7

closed 2023-12-05 by jwellbelove