ETLCPP/etl #788: etl::message_broker does not compile, problem in span.h

tunguskar, 2023-11-14:
Hi there, used the test code from your repo and had the minimal example with the message router which works so far. But when I add the class (and header) for the message_broker from your test code like this:

```cpp
  class Broker : public etl::message_broker
  {
  public:

    Broker()
      : message_broker()
    {
    }

    Broker(etl::message_router_id_t id)
      : message_broker(id)
    {
    }

    using etl::message_broker::receive;

    // Hook incoming messages and translate Message5 to Message4.
    void receive(const etl::imessage& msg) override
    {
      etl::message_broker::receive(msg);
    }
  };
```

I get a synstax error from the compiler from etl::span.h

[image]

tunguskar, 2023-11-14:
I saw that already this failed 

[image]

the error message says on the enum that on the enum there is a missing tag name. Currently Im using the last version of clang. When I remove the header the message is gone.

tunguskar, 2023-11-14:
When I do it like this my example compiles 
[image]

jwellbelove, 2023-11-14:
Does this compile OK for you?

```
#include "etl/message_broker.h"

enum
{
  MESSAGE0,
  MESSAGE1,
  MESSAGE2,
  MESSAGE3,
  MESSAGE4,
  MESSAGE5
};

class Broker : public etl::message_broker
{
public:

  Broker()
    : message_broker()
  {
  }

  Broker(etl::message_router_id_t id)
    : message_broker(id)
  {
  }

  using etl::message_broker::receive;

  // Hook incoming messages and translate Message5 to Message4.
  void receive(const etl::imessage& msg) override
  {
    etl::message_broker::receive(msg);
  }
};

int main()
{
  Broker broker;
}
```

tunguskar, 2023-11-14:
> Broker broker;

No, the first error is this 
[image]

tunguskar, 2023-11-14:
> > Broker broker;
> 
> No, the first error is this [image]

Sorry for the German, it says `Syntax error`

tunguskar, 2023-11-14:
> > > Broker broker;
> > 
> > 
> > No, the first error is this [image]
> 
> Sorry for the German, it says `Syntax error`

[image]

jwellbelove, 2023-11-14:
What version of the ETL are you using?

jwellbelove, 2023-11-14:
What version of clang are you using?

jwellbelove, 2023-11-14:
From the error lines it look like you are compiling for C++11 or above.

It seems to be having a problem with the template function definition:-

```
template <size_t OFFSET, size_t COUNT = etl::dynamic_extent>
ETL_NODISCARD ETL_CONSTEXPR
etl::span<element_type, COUNT != etl::dynamic_extent ? COUNT : Extent - OFFSET> subspan() const ETL_NOEXCEPT
```

jwellbelove, 2023-11-14:
If you have created an `etl::profile.h` or set compiler ETL constants, what are they set to?

tunguskar, 2023-11-14:
> From the error lines it look like you are compiling for C++11 or above.
>
> It seems to be having a problem with the template function definition:-
>
> ```
> template <size_t OFFSET, size_t COUNT = [...]

Clang is 17.0.3
ETL is 20.38.5

I have no profile.h created in my test main I just use: #define ETL_CPP17_SUPPORTED 1

jwellbelove, 2023-11-14:
Ubuntu 22.04
clang version 15.0.7

I've tried to run the complete unit test et for ETL 20.38.5, and the demo code above, but I get no errors.

tunguskar, 2023-11-15:
> Ubuntu 22.04 clang version 15.0.7
> 
> I've tried to run the complete unit test set for ETL 20.38.5, and the demo code above, but I get no errors.

Ok I tried to check if the cpp17 is activated in cmake environment. This seems ok so far.  Have you set the profile.h file? If yes could I get it for a try?

jwellbelove, 2023-11-15:
Ubuntu 22.04
clang version 15.0.7
ETL 20.38.5

**Demo.cpp**
```
#include <iostream>

#include "etl/message_broker.h"

enum
{
  MESSAGE0,
  MESSAGE1, 
  MESSAGE2,
  MESSAGE3,
  MESSAGE4,
  MESSAGE5
};

class Broker : public etl::message_broker
{
public:

  Broker()
    : message_broker()
  {
  }

  Broker(etl::message_router_id_t id)
    : message_broker(id)
  {
  }

  using etl::message_broker::receive;

  void receive(const etl::imessage& msg) override
  {
    etl::message_broker::receive(msg);
  }
};

int main()
{
  std::cout << "Demo\n";

  Broker broker;
}
```

Compiles without errors using this command line.

`clang -I etl/include -lstdc++ -std=c++17 Demo.cpp -o Demo`

jwellbelove, 2023-11-15:
`etl/platform.h` will automatically determine many of the compiler and language properties.

jwellbelove, 2023-11-15:
**CMakeLists.txt**
```
cmake_minimum_required(VERSION 3.5.0)
project(etl_demo_code LANGUAGES CXX)

add_executable(etl_demo
	Demo.cpp
  )

message(STATUS "Compiling for C++17")    
set_property(TARGET etl_demo PROPERTY CXX_STANDARD 17)

if ((CMAKE_CXX_COMPILER_ID MATCHES "MSVC"))
	message(STATUS "Using MSVC compiler")
	target_compile_options(etl_tests
			PRIVATE
			/Zc:__cplusplus
			)
endif ()

if (CMAKE_CXX_COMPILER_ID MATCHES "GNU")
	message(STATUS "Using GCC compiler")
endif ()

if (CMAKE_CXX_COMPILER_ID MATCHES "Clang")
    message(STATUS "Using Clang compiler")
endif ()

target_include_directories(etl_demo
		PRIVATE
		${PROJECT_SOURCE_DIR}etl/include)
```

Invoke with:-
`cmake -DCMAKE_CXX_COMPILER="clang++" ./`

jwellbelove, 2023-11-15:
**Output**
```
-- Compiling for C++17
-- Using Clang compiler
-- Configuring done
-- Generating done
```

tunguskar, 2024-01-27:
> **CMakeLists.txt**
> 
> ```
> cmake_minimum_required(VERSION 3.5.0)
> project(etl_demo_code LANGUAGES CXX)
> 
> add_executable(etl_demo
> 	Demo.cpp
>   )
> 
> message(STATUS "Compiling for C++17")    
> set_property(TARGET etl_demo PROPERTY CXX_STANDARD 17)
> 
> if ((CMAKE_CXX_COMPILER_ID MATCHES "MSVC"))
> 	message(STATUS "Using MSVC compiler")
> 	target_compile_options(etl_tests
> 			PRIVATE
> 			/Zc:__cplusplus
> 			)
> endif ()
> 
> if (CMAKE_CXX_COMPILER_ID MATCHES "GNU")
> 	message(STATUS "Using GCC compiler")
> endif ()
> 
> if (CMAKE_CXX_COMPILER_ID MATCHES "Clang")
>     message(STATUS "Using Clang compiler")
> endif ()
> 
> target_include_directories(etl_demo
> 		PRIVATE
> 		${PROJECT_SOURCE_DIR}/etl/include)
> ```
> 
> Invoke with:- `cmake -DCMAKE_CXX_COMPILER="clang++" ./` `cmake --build .`
> 
> **Output**
> 
> ```
> -- Compiling for C++17
> -- Using Clang compiler
> -- Configuring done
> -- Generating done
> ```
> 
> ```
> [ 50%] Building CXX object CMakeFiles/etl_demo.dir/Demo.cpp.o
> [100%] Linking CXX executable etl_demo
> [100%] Built target etl_demo
> ```

On which machine do you build the code? Is it a windows machine? I thought clang uses in the back the microsoft compiler on a windows machine. And there are some problems I think.

jwellbelove, 2024-01-27:
I'm using clang on Ubuntu 22.04

closed 2024-04-15 by jwellbelove