ETLCPP/etl #796: etl::string_ext without initializing string possible?

basprins, 2023-12-01:
Hi,
Is there any way I can support the following?

I have a `uint8_t *` buffer which is filled with raw bytes after reading from flash (ascii based).
This buffer contains characters I need to replace, so I want to use this buffer in a string. I through the `et::string_ext` would fit my needs.
Lets say I have a buffer of 512, which I can guarantee is only filled with 256 characters.
Then I want to make an etl string_ext for the buffer of 512 bytes (knowing that at a maximum only half of the buffer contains data).
Then I can "freely" start seaching and replacing characters to my need.

But every constructor of `etl_string_ext` also initializes (or assigns) the string, resulting in overwriting the first byte in the string with a 0x0 byte. I explicitly don't want that, I just want to get all the features of etl::string_ext, without tampering with the data. I want etl to only set the size of the string based on the 0x0 terminated string, but leave all preexisting characters alone.

Is this possible? And if so, could you give me a hint, because I'm looking in the wrong places it seems.

jwellbelove, 2023-12-01:
There isn't a way at the moment. I will have to add a flag to the constructor to disable null termination of the buffer. I can send you an interim patch if you like?

basprins, 2023-12-01:
Oh you are fast! 
Yes, that would be fantastic. I sorted it locally by saving and restoring the first byte and uninitialized resize. But a fix in ETL would be great of course.

jwellbelove, 2023-12-01:
Actually, there is a way, I just forgot about it.
You use the constructor this constructor...
`string_ext(const char* text, char* buffer, size_type buffer_size)`
The constructor will recognise that the address of `text` is the same as `buffer` and just set the size of the string to the length of `text`.

basprins, 2023-12-01:
I will be damned. 

> But every constructor of `etl_string_ext` also initializes (or assigns) the string

I would've sworn I tried them all. But I confirm, problem solved! Thank you sir!

jwellbelove, 2023-12-01:
> But every constructor of `etl_string_ext` also initializes (or assigns) the string
If the address of `text` is different from `buffer`, then it will.

basprins, 2023-12-01:
Yeah had a quick look, silly that I missed that.