ETLCPP/etl #797: Undeclared FP_NAN, FP_INFINITE and FP_ZERO

jputcu, 2023-12-05:
With the latest `etl-20.38.7` I get the following compilation errors on `avr-g++-7.3(Microchip)`:
~~~~
C:/Users/joris/.conan2/p/etl187c506cfdd84/p/include/etl/math.h: In function 'constexpr typename etl::enable_if<etl::is_floating_point<T>::value, bool>::type etl::is_nan(T)':
C:/Users/joris/.conan2/p/etl187c506cfdd84/p/include/etl/math.h:58:33: error: 'FP_NAN' was not declared in this scope
     return fpclassify(value) == FP_NAN;
                                 ^~~~~~
C:/Users/joris/.conan2/p/etl187c506cfdd84/p/include/etl/math.h:58:33: note: suggested alternative: 'PINA'
     return fpclassify(value) == FP_NAN;
                                 ^~~~~~
                                 PINA
C:/Users/joris/.conan2/p/etl187c506cfdd84/p/include/etl/math.h: In function 'constexpr typename etl::enable_if<etl::is_floating_point<T>::value, bool>::type etl::is_infinity(T)':
C:/Users/joris/.conan2/p/etl187c506cfdd84/p/include/etl/math.h:89:33: error: 'FP_INFINITE' was not declared in this scope
     return fpclassify(value) == FP_INFINITE;
                                 ^~~~~~~~~~~
C:/Users/joris/.conan2/p/etl187c506cfdd84/p/include/etl/math.h:89:33: note: suggested alternative: 'INFINITY'
     return fpclassify(value) == FP_INFINITE;
                                 ^~~~~~~~~~~
                                 INFINITY
C:/Users/joris/.conan2/p/etl187c506cfdd84/p/include/etl/math.h: In function 'constexpr typename etl::enable_if<etl::is_floating_point<T>::value, bool>::type etl::is_zero(T)':
C:/Users/joris/.conan2/p/etl187c506cfdd84/p/include/etl/math.h:121:33: error: 'FP_ZERO' was not declared in this scope
     return fpclassify(value) == FP_ZERO;
                                 ^~~~~~~
~~~~

I already use the following compiler flags:
`-DETL_NO_CPP_NAN_SUPPORT -DETL_NO_HUGE_VAL_SUPPORT -DETL_NO_STL -DETL_TARGET_DEVICE_GENERIC -DETL_TARGET_OS_NONE`

jwellbelove, 2023-12-05:
Does this work for you?

[etl-797-Undeclared-FP_NAN,-FP_INFINITE-and-FP_ZERO.patch](https://github.com/ETLCPP/etl/files/13557610/etl-797-Undeclared-FP_NAN.-FP_INFINITE-and-FP_ZERO.patch)

jputcu, 2023-12-05:
Yes, that fixes the issue

jwellbelove, 2023-12-05:
Fixed 20.38.8

closed 2023-12-05 by jwellbelove