ETLCPP/etl #802: Fuzz testing with google/oss-fuzz

silvergasp, 2023-12-10:
Hey ETL Team,

I hope this message finds you well. I've been following along with ETL for some time now, and have integrated it into a number of projects. I'd like to suggest and champion an effort to set up some basic fuzz-testing and combine it with google/oss-fuzz for continuous fuzzing. I'm fully aware that you are very busy people and I don't want to overload your review/maintenance capacity by introducing too many new ideas. Is this a bad time to discuss potential security/reliability improvements? 

If your not familiar with fuzzing or oss-fuzz I've included a few brief notes below.

#### **Benefits of Fuzz-Testing**

- **Dynamic Code Testing**: Fuzz-testing challenges systems with unexpected data, aiming to identify vulnerabilities. It’s akin to an exhaustive stress-test for the code.
- **Detecting Hidden Vulnerabilities**: It can uncover potential weaknesses that may not be evident in routine tests.
- **Continuous and Automated Testing**: With tools like Google’s OSS-Fuzz, fuzz-testing can be automated, running continuously on distributed systems, ensuring daily resilience checks.

#### **Google/oss-fuzz for Continuous Fuzzing**

- **Automated Fuzzing**: OSS-Fuzz undertakes comprehensive fuzz-testing daily on a distributed cluster.
- **Security Boost**: It provides enhanced security measures free of cost, thanks to Google’s backing.
- **Detailed Reporting**: OSS-Fuzz offers exhaustive reports in case of detected anomalies, enabling effective action.

I’d be more than happy to lead the effort in integrating fuzz testing with ETL and assist in any way required.

As a proof of concept I created a couple of super simple fuzz harnesses for the for the some of the hashers and also string functions in #801.

silvergasp, 2024-09-01:
@jwellbelove friendly ping. I'm about to take another pass over this, but I'm a little puzzled by what the current state is. Specifically it looks like you changed the source branch for #801 from main/master to '801-fuzz-Add-fuzz-harnesses-for-string-crc-apis' and then merged it into that branch. Is there something that you wanted changed in the original PR before it made it into the main branch?

I guess I feel like I'm missing something important as I've not seen a workflow similar to this.

Once it's merged into the main branch it'll be a relatively minimal amount of effort for me to integrate this into oss-fuzz, if that's something that you are still interested in.

jwellbelove, 2024-09-01:
I like to keep pull requests and issues on their own branch until I am happy that they are ready to merge into 'master'. The branch I change them to is based on master.

silvergasp, 2024-09-01:
Ah ok understood.

I'll quickly draft up an integration with oss-fuzz. Then if/when you have the bandwidth for it feel free to ping me and I'll go over #801 with you and answer any questions and make changes as needed.