ETLCPP/etl #808: new etl::bitset does not compile for 32bit compilation

TG-SAG, 2023-12-18:
I'm currently switching our project from the **legacy** `etl::bitset<>` to the **new** `etl::bitset<>`. But I have problems compiling the unit tests for a 32bit platform (we explicitly instantiate a selection of sizes to make sure they will compile): https://gcc.godbolt.org/z/Yrh7G9rMd

`template class etl::bitset<64>; `

The above line will produce the follwing error:

```
/opt/compiler-explorer/libs/etl/include/etl/private/bitset_new.h: In instantiation of 'constexpr typename etl::enable_if<etl::is_integral<TIterator>::value, T>::type etl::bitset<Active_Bits, TElement, false>::value() const [with T = long unsigned int; unsigned int Active_Bits = 64; TElement = char; typename etl::enable_if<etl::is_integral<TIterator>::value, T>::type = long unsigned int]':
/opt/compiler-explorer/libs/etl/include/etl/private/bitset_new.h:2035:34:   required from 'constexpr long unsigned int etl::bitset<Active_Bits, TElement, false>::to_ulong() const [with unsigned int Active_Bits = 64; TElement = char]'
<source>:4:21:   required from here
/opt/compiler-explorer/libs/etl/include/etl/private/bitset_new.h:2025:7: error: static assertion failed: Type too small
 2025 |       ETL_STATIC_ASSERT((sizeof(T) * CHAR_BIT) >= (Number_Of_Elements * Bits_Per_Element), "Type too small");
      |       ^~~~~~~~~~~~~~~~~
/opt/compiler-explorer/libs/etl/include/etl/private/bitset_new.h:2025:7: note: the comparison reduces to '(32 >= 64)'
```
I will seems to work for a 64bit compilation: https://gcc.godbolt.org/z/h5T87Eof1

Before I start fixing the problem I wanted to ask if someone already has a fix for this (because I shouldn't be that uncommon to have a 32bit compilation)

jwellbelove, 2023-12-18:
I'm not sure why the template function `value()` is being instantiated when the compiler has absolutely no idea what the requested return type actually might be!

Where exactly is `T` being defined in the declaration `template class etl::bitset<64>;` for the template member function below?

Why is it assuming `T = long unsigned int;` if value() is never invoked?

```
template <typename T>
ETL_CONSTEXPR14
typename etl::enable_if<etl::is_integral<T>::value, T>::type
  value() const ETL_NOEXCEPT
{
  ETL_STATIC_ASSERT(etl::is_integral<T>::value, "Only integral types are supported");
  ETL_STATIC_ASSERT((sizeof(T) * CHAR_BIT) >= (Number_Of_Elements * Bits_Per_Element), "Type too small");

  return ibitset.template value<T>(buffer, Number_Of_Elements);
}
```

jwellbelove, 2023-12-19:
Ah yes, I see. The error only mentioned value(). I forgot about those other member functions.
Adding `enable_if` would be pretty simple to add; I may add it while I'm doing the current changes.

As well as the requested extract functions I'm currently working on, I'm looking at reducing the amount of code generated for varying bit lengths that fit into a single integer.

For example, `etl::bitset<5, uint8_t>`, `etl::bitset<7, uint8_t>` & `etl::bitset<8, uint8_t>` would cause three almost identical classes to be generated. I am looking at moving a lot of the guts of the top level classes into a bit length independent implementation.

jwellbelove, 2023-12-19:
Actually, a `static_assert` would be easier than `enable_if`.

Also, to address the code size issues, I'm thinking of mandating that the element type **must** be unsigned.

jwellbelove, 2024-04-21:
Fixed 20.38.11

closed 2024-04-21 by jwellbelove