ETLCPP/etl #809: std::bit_cast equivalent?

jaskij, 2023-12-19:
While I'm personally happy to use ARM's latest GNU toolchain, a lot of ETL users don't have C++20 yet. Casting between types (such as floating point numbers to an array of uint8_t) in a standard compliant way before C++20 has it's pitfalls. Porting [std::bit_cast](https://en.cppreference.com/w/cpp/numeric/bit_cast) might be a good idea. Sadly, I don't believe it's possible to it be `constexpr`.

jwellbelove, 2023-12-19:
The ETL already has one.
`etl::bit_cast` in `bit.h`

If you are bit casting between same sized integrals, then `etl::bit_cast` is `constexpr` (from C++14)

jaskij, 2023-12-19:
Ah, sorry, couldn't find it in the documentation. Great news.

closed 2023-12-19 by jaskij

jwellbelove, 2023-12-19:
I'll check and update the documentation page.

jaskij, 2023-12-21:
Having checked the documentation once again, it's probably my unfamiliarity with how your documentation is laid out. To be frank, it's been a while since I have used external libraries (only recently started doing it on microcontrollers), so it's entirely on me.

jwellbelove, 2023-12-21:
The documentation pages need a update at some point to make the features in the ETL more 'discoverable'.

jaskij, 2023-12-21:
I think what you're missing is library/section overview pages. Looking at cppreference, compare the page for the [numerics library](https://en.cppreference.com/w/cpp/numeric) to the one for the [`<bit>`](https://en.cppreference.com/w/cpp/header/bit) header. You have the latter, but are missing the former.

jwellbelove, 2023-12-21:
The documentation pages are created manually at the moment, but the project has grown so much over the last nine years (!!!) that I will need to look at some of the auto generation options.
I used to use Doxygen in a previous job, but there are more options nowadays.

jaskij, 2023-12-21:
> The documentation pages are created manually at the moment, but the project has grown so much over the last nine years (!!!) that I will need to look at some of the auto generation options.

Nine years? Woah. This is a great library though, so no wonder. And writing the documentation by hand is in no way the correct solution here.

> I used to use Doxygen in a previous job, but there are more options nowadays.

Yup, personally I always look for tools that can integrate this with [Asciidoctor](https://asciidoctor.org/), or other good writing system. Markdown, while fine for short text like here, is really insufficient for longer technical documentation. And just now I have noticed [AsciiDoxy](https://www.asciidoxy.org/), which combines good ol' Doxygen with Asciidoctor.

It's definitely an area I'd love to help with, as I want to look into those tools myself anyway.