ETLCPP/etl #819: etl::optional requires that T be "pod"

jovere, 2024-01-15:
In the etl::optional code, it is required that the type be `is_pod`, aka. "plain old data". Why is this? Why can't I have an optional of a more complex data type?

For reference:
https://github.com/ETLCPP/etl/blob/0f1840a70d11a3d317afa55b8361829b93c8eeb8/include/etl/optional.h#L107

jwellbelove, 2024-01-16:
ETL's optional can used with both OD and complex types.
There are two specialisations:-
One where the type is a POD. `etl::optional<T, true>`
The other where the type is not POD. `etl::optional<T, false>`

jwellbelove, 2024-01-16:
`etl::optional<int>` instantiates `etl::optional<T, true>`
`etl::optional<std::string>` instantiates `etl::optional<T, false>`

jovere, 2024-01-16:
My apologies,
It appears that I was incorrect in the reading of the error. I plugged it into godbolt (https://godbolt.org/z/PE91sxhaz) and figured out the actual error.

It won't let me return an optional value of the class itself from a member function, since it is forward declared. This is an issue, since we have a static member function to create an object from a string. I've also shown that it works fine with the standard library version. I have also tried boost and it works as expected. The code on godbolt (https://godbolt.org/z/PE91sxhaz) is shown below for posterity:

```
#include <string>
#include <optional>
#include "etl/optional.h"

class etl_optional_dataType{    

public:
    static etl::optional<etl_optional_dataType> createItemFromString(std::string data) // Won't compile
    {
        return {};
    }
};

class std_optional_dataType{
public:
    static std::optional<std_optional_dataType> createItemFromString(std::string data) // Compiles
    {
        return {};
    }
};
```

jwellbelove, 2024-01-17:
Looks like I may have to do the POD optimisations differently.

jwellbelove, 2024-04-21:
Fixed 20.38.11

closed 2024-04-21 by jwellbelove