ETLCPP/etl #823: Multiple variable shadowing warnings in private/bitset_new.h

digimokan, 2024-01-22:
When compiling with -Wshadow, there are multiple warnings in private/bitset_new.h.

There is one warning for the `lsb` variable shadowing:

https://github.com/ETLCPP/etl/blob/master/include/etl/private/bitset_new.h#L643

And the rest of the warnings come from `value` variable shadowing:

https://github.com/ETLCPP/etl/blob/master/include/etl/private/bitset_new.h#L190

https://github.com/ETLCPP/etl/blob/master/include/etl/private/bitset_new.h#L532

https://github.com/ETLCPP/etl/blob/master/include/etl/private/bitset_new.h#L766

https://github.com/ETLCPP/etl/blob/master/include/etl/private/bitset_new.h#L974

https://github.com/ETLCPP/etl/blob/master/include/etl/private/bitset_new.h#L1038

https://github.com/ETLCPP/etl/blob/master/include/etl/private/bitset_new.h#L1048

https://github.com/ETLCPP/etl/blob/master/include/etl/private/bitset_new.h#L1855

https://github.com/ETLCPP/etl/blob/master/include/etl/private/bitset_new.h#L1926

https://github.com/ETLCPP/etl/blob/master/include/etl/private/bitset_new.h#L2642

https://github.com/ETLCPP/etl/blob/master/include/etl/private/bitset_new.h#L2651

https://github.com/ETLCPP/etl/blob/master/include/etl/private/bitset_new.h#L2760

https://github.com/ETLCPP/etl/blob/master/include/etl/private/bitset_new.h#L3530

https://github.com/ETLCPP/etl/blob/master/include/etl/private/bitset_new.h#L3520

https://github.com/ETLCPP/etl/blob/master/include/etl/private/bitset_new.h#L3641

jwellbelove, 2024-01-23:
I'm currently refactoring `bitset_new.h`, so I'll look at these changes while I'm at it.

jwellbelove, 2024-04-21:
Fixed 20.38.11

closed 2024-04-21 by jwellbelove