ETLCPP/etl #824: Feature Request `etl::message_router` virtual inheritance

edobez, 2024-01-23:
Similar to #384, would be nice to have a virtual inheritance or templated base class for `etl::message_router` as well.
This would allow to create custom interface classes for certain kinds of routers.

For what I saw having a templated base class as for `etl::message` won't work with the pre-C++17 implementation, because supposedly the template argument for the base class should have a default value and that requires a specific ordering of such arguments.

Probably the virtual inheritance solution would work. Why was it discarded for the related issue?

jwellbelove, 2024-01-27:
_For what I saw having a templated base class as for `etl::message` won't work with the pre-C++17 implementation_

Did you mean `etl::message_router`?

edobez, 2024-01-27:
I wasn’t clear enough with that statement.
I wanted to say that the solution for selecting a custom base class, existing for `etl::message`, which provides an optional template argument for specifying the base interface, won’t work in the case of `etl::message_router`, because the latter has already many optional template arguments (in the pre C++17 version).

—
Edoardo