ETLCPP/etl #835: Compiler error with etl::variant

mike919192, 2024-01-31:
Hi

I am getting compiler errors when I try to add etl variant to my project.  I am using gcc-arm-none-eabi version 12.2 and standard c++20.

My usage is simple etl::variant<int, float>

Sorry for the long list but here are all the errors:

```
All errors in variant_legacy.h
'<declaration error>' is not a template [-fpermissive]
'<declaration error>' is not a template [-fpermissive]
'<declaration error>' is not a template [-fpermissive]
'<declaration error>' is not a template [-fpermissive]
'<declaration error>' is not a template [-fpermissive]
'<declaration error>' is not a template [-fpermissive]
'<declaration error>' is not a template [-fpermissive]
'<declaration error>' is not a template [-fpermissive]
'R4' was not declared in this scope; did you mean 'T4'?
'R4' was not declared in this scope; did you mean 'T4'?
'R4' was not declared in this scope; did you mean 'T4'?
'R4' was not declared in this scope; did you mean 'T4'?
'R4' was not declared in this scope; did you mean 'T4'?
'R5' was not declared in this scope; did you mean 'T5'?
'R5' was not declared in this scope; did you mean 'T5'?
'R5' was not declared in this scope; did you mean 'T5'?
'R5' was not declared in this scope; did you mean 'T5'?
'R6' was not declared in this scope; did you mean 'T6'?
'R6' was not declared in this scope; did you mean 'T6'?
'R6' was not declared in this scope; did you mean 'T6'?
'R7' was not declared in this scope; did you mean 'T7'?
'R7' was not declared in this scope; did you mean 'T7'?
'R8' was not declared in this scope; did you mean 'T8'?
'virtual void etl::legacy::variant<T1, T2, T3, T4, T5, T6, T7, T8>::reader_type<<anonymous> >::read(int)' cannot be overloaded with 'virtual void etl::legacy::variant<T1, T2, T3, T4, T5, T6, T7, T8>::reader_type<<anonymous> >::read(int)'
'virtual void etl::legacy::variant<T1, T2, T3, T4, T5, T6, T7, T8>::reader_type<<anonymous> >::read(int)' cannot be overloaded with 'virtual void etl::legacy::variant<T1, T2, T3, T4, T5, T6, T7, T8>::reader_type<<anonymous> >::read(int)'
'virtual void etl::legacy::variant<T1, T2, T3, T4, T5, T6, T7, T8>::reader_type<<anonymous> >::read(int)' cannot be overloaded with 'virtual void etl::legacy::variant<T1, T2, T3, T4, T5, T6, T7, T8>::reader_type<<anonymous> >::read(int)'
'virtual void etl::legacy::variant<T1, T2, T3, T4, T5, T6, T7, T8>::reader_type<<anonymous> >::read(int)' cannot be overloaded with 'virtual void etl::legacy::variant<T1, T2, T3, T4, T5, T6, T7, T8>::reader_type<<anonymous> >::read(int)'
'virtual void etl::legacy::variant<T1, T2, T3, T4, T5, T6, T7, T8>::reader_type<<anonymous> >::read(int)' cannot be overloaded with 'virtual void etl::legacy::variant<T1, T2, T3, T4, T5, T6, T7, T8>::reader_type<<anonymous> >::read(int)'
'virtual void etl::legacy::variant<T1, T2, T3, T4, T5, T6, T7, T8>::reader_type<<anonymous> >::read(int)' cannot be overloaded with 'virtual void etl::legacy::variant<T1, T2, T3, T4, T5, T6, T7, T8>::reader_type<<anonymous> >::read(int)'
'virtual void etl::legacy::variant<T1, T2, T3, T4, T5, T6, T7, T8>::reader_type<<anonymous> >::read(int)' cannot be overloaded with 'virtual void etl::legacy::variant<T1, T2, T3, T4, T5, T6, T7, T8>::reader_type<<anonymous> >::read(int)'
request for member 'read' in 'r', which is of non-class type 'etl::legacy::variant<T1, T2, T3, T4, T5, T6, T7, T8>::reader' {aka 'int'}
request for member 'read' in 'r', which is of non-class type 'etl::legacy::variant<T1, T2, T3, T4, T5, T6, T7, T8>::reader' {aka 'int'}
request for member 'read' in 'r', which is of non-class type 'etl::legacy::variant<T1, T2, T3, T4, T5, T6, T7, T8>::reader' {aka 'int'}
request for member 'read' in 'r', which is of non-class type 'etl::legacy::variant<T1, T2, T3, T4, T5, T6, T7, T8>::reader' {aka 'int'}
request for member 'read' in 'r', which is of non-class type 'etl::legacy::variant<T1, T2, T3, T4, T5, T6, T7, T8>::reader' {aka 'int'}
request for member 'read' in 'r', which is of non-class type 'etl::legacy::variant<T1, T2, T3, T4, T5, T6, T7, T8>::reader' {aka 'int'}
request for member 'read' in 'r', which is of non-class type 'etl::legacy::variant<T1, T2, T3, T4, T5, T6, T7, T8>::reader' {aka 'int'}
request for member 'read' in 'r', which is of non-class type 'etl::legacy::variant<T1, T2, T3, T4, T5, T6, T7, T8>::reader' {aka 'int'}
template argument 1 is invalid
template argument 1 is invalid
template argument 1 is invalid
template argument 1 is invalid
template argument 1 is invalid
type/value mismatch at argument 1 in template parameter list for 'template<class T> struct etl::parameter_type'
type/value mismatch at argument 1 in template parameter list for 'template<class T> struct etl::parameter_type'
type/value mismatch at argument 1 in template parameter list for 'template<class T> struct etl::parameter_type'
wrong number of template arguments (8, should be 1)
wrong number of template arguments (8, should be 1)
wrong number of template arguments (8, should be 1)
wrong number of template arguments (8, should be 1)
wrong number of template arguments (8, should be 1)
wrong number of template arguments (8, should be 1)
wrong number of template arguments (8, should be 1)
wrong number of template arguments (8, should be 1)
```

Some small troubleshooting steps I have done one at a time:

- Replace etl::variant with std::variant and no errors
- Compile the same code for x86_64 Linux gcc 11.4 and no errors
- On a whim I also tried to remove the include for variant_legacy.h from variant.h like this.  Also no errors 
```
#ifndef ETL_VARIANT_INCLUDED
#define ETL_VARIANT_INCLUDED

#include "platform.h"

#if !ETL_USING_CPP11 || defined(ETL_USE_LEGACY_VARIANT)
  #include "private/variant_legacy.h"   // etl::variant
#else
  //#include "private/variant_legacy.h"   // etl::legacy::variant
  #include "private/variant_variadic.h" // etl::variant
#endif

#endif
```

If there is any other information I can provide, let me know.  Thanks

jwellbelove, 2024-01-31:
I'll take a look at the errors, but `etl::variant_new` is to be preferred over `etl:: variant_legacy` as it has better conformance with the API of `std::variant`.

jwellbelove, 2024-01-31:
The one selected is decided in `variant.h`.

jwellbelove, 2024-01-31:
That's odd, the inclusion of the two headers should be mutually exclusive.

mike919192, 2024-01-31:
This is the unmodified variant.h.  If the variant_variadic is selected, the legacy is also included but placed in a separate namespace.

```
#ifndef ETL_VARIANT_INCLUDED
#define ETL_VARIANT_INCLUDED

#include "platform.h"

#if !ETL_USING_CPP11 || defined(ETL_USE_LEGACY_VARIANT)
  #include "private/variant_legacy.h"   // etl::variant
#else
  #include "private/variant_legacy.h"   // etl::legacy::variant
  #include "private/variant_variadic.h" // etl::variant
#endif

#endif
```

mike919192, 2024-01-31:
I also checked and the errors are triggered with only `#include "etl/variant.h"` in the main.cpp.  No instance of an actual etl::variant is needed to trigger the errors.

closed 2024-01-31 by mike919192

jwellbelove, 2024-01-31:
Ah macros, isn't it fun when they apply themselves to all scopes!

I had a strange error many years ago with Win32 where my class had a `GetMessage` member function and the compiler was complaining that it couldn't find `GetMessageA`.
Turned out that the windows header had a macro that defined `GetMessage` as either `GetMessageA` or `GetMessageW` depending of the use of ascii or wide char types.