ETLCPP/etl #837: Google GMock and etl::delegate work not together

NicoMayer, 2024-02-07:
Hello, I use ETL in an embedded project. GoogleTest is used for software testing. There is now a major problem with etl::delegate and gmock when implementing a test.

### The Test:
```cpp
#include "etl/delegate.h"
#include "gmock/gmock.h"

using namespace std;
using namespace testing;

using Callback = etl::delegate<void(void)>;

class IHardwareSensor {
 public:
  virtual int attachSensorCallback(Callback callback) = 0;
};

class HardwareSensorMock : public IHardwareSensor {
 public:
  MOCK_METHOD(int, attachSensorCallback, (Callback));
};

class Sensor {
 public:
  Sensor(IHardwareSensor &sensor) { sensor.attachSensorCallback(Callback::create<Sensor, &Sensor::handleNewSensorValue>(*this)); }

 private:
  void handleNewSensorValue() {}
};

TEST(SensorTest, init) {
  HardwareSensorMock mockedHardwareSensor;

  Callback newSensorValueCallback;

  EXPECT_CALL(mockedHardwareSensor, attachSensorCallback(An<Callback>())).Times(Exactly(1)).WillOnce(DoAll(SaveArg<0>(&newSensorValueCallback), Return(0)));

  Sensor sensor(mockedHardwareSensor);
}
```

### The Compiler Output (In German)
```
[build] Microsoft (R) C/C++-Optimierungscompiler Version 19.29.30038.1 für x86
[build] Copyright (C) Microsoft Corporation. Alle Rechte vorbehalten.
[build] 
[build] C:\Projekte\xxxxxxxx\checkout\software\application\xxxxxxxx\_libs\etl\include\etl\private/delegate_cpp11.h(566): error C2039: "()" ist kein Member von "std::tuple<_Ty &&>".
[build]         with
[build]         [
[build]             _Ty=etl::delegate<void (void)>
[build]         ]
[build] C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30037\include\tuple(210): note: Siehe Deklaration von "std::tuple<_Ty &&>"
[build]         with
[build]         [
[build]             _Ty=etl::delegate<void (void)>
[build]         ]
[build] C:\Projekte\xxxxxxxx\checkout\software\application\xxxxxxxx\_libs\etl\include\etl\private/delegate_cpp11.h(123): note: Siehe Verweis auf die gerade kompilierte Instanziierung "TReturn etl::delegate<void (void)>::const_lambda_stub<TLambda>(void *)" der Funktions-Vorlage.
[build]         with
[build]         [
[build]             TReturn=void,
[build]             TLambda=std::tuple<etl::delegate<void (void)> &&>
[build]         ]
[build] C:\Projekte\xxxxxxxx\checkout\software\application\xxxxxxxx\_libs\etl\include\etl\private/delegate_cpp11.h(124): note: Siehe Verweis auf die gerade kompilierte Instanziierung "TReturn etl::delegate<void (void)>::const_lambda_stub<TLambda>(void *)" der Funktions-Vorlage.
[build]         with
[build]         [
[build]             TReturn=void,
[build]             TLambda=std::tuple<etl::delegate<void (void)> &&>
[build]         ]
[build] C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30037\include\tuple(170): note: Siehe Verweis auf die gerade kompilierte Instanziierung "etl::delegate<void (void)>::delegate<std::tuple<_Ty &&>,void>(const TLambda &)" der Funktions-Vorlage.
[build]         with
[... 163 lines cut ...]
### The quick fix
```cpp
    //*************************************************************************
    // Construct from lambda or functor.
    //*************************************************************************
    template <typename TLambda, typename = etl::enable_if_t<etl::is_class<TLambda>::value && !etl::is_same<etl::delegate<TReturn(TParams...)>, TLambda>::value && std::is_invocable_r<TReturn, TLambda, TParams...>::value, void>>
    ETL_CONSTEXPR14 delegate(TLambda& instance)
    {
      assign((void*)(&instance), lambda_stub<TLambda>);
    }

    //*************************************************************************
    // Construct from const lambda or functor.
    //*************************************************************************
    template <typename TLambda, typename = etl::enable_if_t<etl::is_class<TLambda>::value && !etl::is_same<etl::delegate<TReturn(TParams...)>, TLambda>::value && std::is_invocable_r<TReturn, TLambda, TParams...>::value, void>>
    ETL_CONSTEXPR14 delegate(const TLambda& instance)
    {
      assign((void*)(&instance), const_lambda_stub<TLambda>);
    }
```

So the question now is how I should deal with the problem in the long term.

Best regards and sorry for my bad English
Nico

jwellbelove, 2024-02-15:
Unfortunately, many type traits are next to impossible for the ETL to reverse engineer due to the reliance on the compiler's 'insider knowledge'.
The only way to use your fix would be to only enable it if the STL is being used.
`#if ETL_USING_STL`

ilya-a1, 2024-07-08:
The following does work for unittests:
 EXPECT_CALL(mockedHardwareSensor, attachSensorCallback(_)).Times(Exactly(1)).WillOnce( 
   Invoke( [&newSensorValueCallback](auto& cb){ newSensorValueCallback = Callback::create(cb); return 0; } 
 );

closed 2025-07-20 by jwellbelove