ETLCPP/etl #841: Change push to push_back on containers, so they can be used with std::back_inserter

DavidWild02, 2024-02-12:
this code:
`std::copy_n(reinterpret_cast<const std::byte *>(&data), std::back_inserter(binaryBuffer));`
does not work, because the `etl::circular_buffer` has no member `push_back(__value)`.

It would be nice implementing this. So that the container can be used with functions of the standard library.

jwellbelove, 2024-02-13:
The ETL tries to adhere to STL API conventions to maintain compatibility as much as possible, and so ETL containers like queues and stacks use `push()` and not `push_back()`. As `etl::circular_buffer` is essentially a queue it has a `push()`.

A better alternative may be to create an `etl::push_inserter()` that calls the container's `push()`.

jwellbelove, 2024-04-21:
Resolved 20.38.11

closed 2024-04-21 by jwellbelove