ETLCPP/etl #844: etl::type_def - constexpr instance

dfalk-solve, 2024-02-15:
My question here is, if it would be possible to enhance etl::type_def to allow an constexpr instance that just allows reading the value, which could be helpful for a private member definition within a class.

Currently I do the definitions using it the following way, e.g.

Definition in a common header file:
ETL_TYPEDEF(uint32_t, milliSecond_t);   ///< [ms]   Time measurement (1ms = 0.001s)

And in a private section of a class:
static constexpr milliSecond_t::value_type DefaultTimeout { 10 };

So I define a constant for the "::value_type" which is the underlaying 
data type, which is uint32_t And what I would prefer is to define a constant value for the strong type, in this case "milliSecond_t"

static constexpr milliSecond_t DefaultTimeout { 10 };

Would this be possible if defining a constexpr constructor and a "get()" function? If yes, could this be a possible enhancement for one of the next versions of ETL?

drewr95, 2024-02-19:
I planned on working on this (or if John or someone else beat me to it), see my issue here about the same subject: https://github.com/ETLCPP/etl/issues/838

jwellbelove, 2024-04-21:
Fixed 20.38.11

closed 2024-04-21 by jwellbelove