ETLCPP/etl #845: `message_packet` missing a copy constructor

opcod3, 2024-02-15:
Since commit 047fc3eb906c699d052bcb8dfd32f9f0c8c269e4 pushing a `message_packet` in a `queue_mpmc_mutex` results in a compiler error.

It appears `message_packet` is now missing an lvalue reference copy constructor. 

The error is the following
```
etl/queue_mpmc_mutex.h: In instantiation of 'bool etl::iqueue_mpmc_mutex<T, MEMORY_MODEL>::push_implementation(const_reference) [with T = etl::message_packet<...>; unsigned int MEMORY_MODEL = 2; const_reference = const etl::message_packet<...>&]':
queue_mpmc_mutex.h:162:40:   required from 'bool etl::iqueue_mpmc_mutex<T, MEMORY_MODEL>::push(const_reference) [with T = etl::message_packet<...>; unsigned int MEMORY_MODEL = 2; const_reference = const etl::message_packet<...>&]'
my-file.cpp:   required from here
etl/queue_mpmc_mutex.h:429:9: error: use of deleted function 'constexpr etl::message_packet<...>::message_packet(const etl::message_packet<...>&)'
  429 |         ::new (&p_buffer[write_index]) T(value);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

I've replaced the whole template arguments of message_packet for clarity.

The call that produces the error is the following
```
_actionQueue.push(action);
```
adding `etl::move` like this: `_actionQueue.push(etl::move(action))` fixes the error

jwellbelove, 2024-02-15:
Are you using C++17 or later?

opcod3, 2024-02-15:
Yes, and compiling with `gcc version 12.2.1 20221205 (Arm GNU Toolchain 12.2.Rel1 (Build arm-12.24))`

jwellbelove, 2024-04-21:
Fixed 20.38.11

closed 2024-04-21 by jwellbelove