ETLCPP/etl #847: Add has_active_timer to callback_timer

itavero, 2024-02-22:
We are using `time_to_next` in our code to know when we should call `tick` again if no other events in our system have taken place.
However, you need to know if any timers are active before calling this method.

I'd suggest to add a method that expose the `has_active_timer` check (e.g. `!active_list.empty()`), which is also done in the current `tick` implementation.
Alternatively/additionally, perhaps `time_to_next` should return `0` if there is no active timer (as a `delta` value of `0` would never be in the list, so it's a safe "special" value to use for this case).

itavero, 2024-03-14:
@jwellbelove Any particular reason you made `etl::timer::interval::No_Active_Interval` equal to `0xFFFFFFFFUL` instead of `0`?

jwellbelove, 2024-03-14:
'Impossible' interval number

itavero, 2024-04-11:
Sorry for the late reply. My point was more that `0` is also a "real" impossible value for interval (as the timer would have already been served in that case).

jwellbelove, 2024-04-11:
It shouldn't really matter what the value of `etl::timer::interval::No_Active_Interval` is, as any code that checks for this should be comparing against the constant name, and not a literal value.

itavero, 2024-04-11:
That's true. However, after analyzing the latest released code, I noticed with that version just checking if the result of `time_to_next` is `0` already tells you that there's no active timers.
By using the same value in this update, the result remains the "same".

But indeed, other than that, I don't really care about the exact value either.

jwellbelove, 2024-04-11:
That's a fair point.

jwellbelove, 2024-04-21:
Fixed 20.38.11

closed 2024-04-21 by jwellbelove