ETLCPP/etl #859: `etl::variant` always has a non-trivial destructor

HumblePresent, 2024-03-11:
With the current implementation of `etl::variant`, any instance will always have a non-trivial destructor as defined [here](https://github.com/ETLCPP/etl/blob/master/include/etl/private/variant_variadic.h#L672) and [here](https://github.com/ETLCPP/etl/blob/master/include/etl/private/variant_legacy.h#L230). According to the [cppreference page](https://en.cppreference.com/w/cpp/utility/variant/%7Evariant) for `std::variant`, its destructor is trivial if `std::is_trivially_destructible_v<T>` is `true` for all variant types. The fact that `etl::variant` always has a non-trivial destructor, regardless of the variant types, prevents it and any derivatives like `etl::expected` from being used in functions marked `constexpr`. I am wondering if `etl::variant` could be specialized for cases where all variant types are trivially destructible, so that it behaves the same as `std::variant`.

jwellbelove, 2024-03-11:
That functionality would depend on an implementation of `is_trivially_destructible<T>` being available.
Some of the type traits in the STL are impossible to implement in pure C++. Therefore the ETL would have to depend on either the STL being used, or the relevant intrinsics being available for the development compiler.
This is one of the more complex parts of the ETL to manage. It's something I hope to refactor at some point in the future.

HumblePresent, 2024-03-11:
I didn't realize `is_trivially_destructible<T>` is dependent on compiler intrinsics. I understand that is a difficult thing to manage in a cross-platform library like this. Anyway, thank you for the quick response and maintenance of this library in general. It is much appreciated. Feel free to close this issue if you feel it's more of a "wishlist" type request at this time.

drewr95, 2024-03-12:
I came across this issue as well hoping to store an array of etl::variants in ROM via constexpr.  Alas I couldn't because of it being non-trivially destructible.  

Likewise, thank you for the maintenance of this library.

closed 2024-04-15 by jwellbelove