ETLCPP/etl #864: cyclic_value first/last methods

Sn1cket, 2024-03-14:
Hello,
I'm using the cyclic_value like this:

```
typedef etl::cyclic_value<uint8_t, 0, 23> Hours;
```

So my limits are known by compile time. Sometimes I have the need to access the first/last values of my type without defining a variable. 
Right now this isn't possible because the first/last methods are instance methods and I get this error from the compiler.
```
a nonstatic member reference must be relative to a specific object
```

Wouldn't it be a nice addition to the cyclic_value template variant that cannot change first/last during runtime, to make the first/last methods `static const` or even `constexpr`?

Best regards

jwellbelove, 2024-03-14:
That sounds a good idea.

jwellbelove, 2024-04-21:
Added 20.38.11

closed 2024-04-21 by jwellbelove