ETLCPP/etl #868: etl::variant is missing comparison operators

relativityspace-jsmith, 2024-03-19:
Hi!  Thanks for all the work on this library!
We have had good luck using etl::variant (variadic version) on my project.  However, today I noticed that it isn't comparable, e.g. comparing two variant values with operator== always produces a compiler error even if the types are comparable.  std::variant [does implement](https://en.cppreference.com/w/cpp/utility/variant/operator_cmp) this feature, and it would be really awesome if etl::variant could as well!

Here's my naive attempt at an implementation (it complies at least):
```cpp
namespace etl
{
template <class... Types>
constexpr bool operator==(const etl::variant<Types...>& lhs, const etl::variant<Types...>& rhs)
{
    // Are the held types the same?
    if (lhs.index() != rhs.index()) return false;

    // Both valueless?
    if (lhs.valueless_by_exception()) return true;

    // Now check if the contained values compare as equal
    return etl::visit([&](auto const & lhs_downcasted)
    {
        return lhs_downcasted == etl::get<decltype(lhs_downcasted)>(rhs);
    }, lhs);
}
}
```

jwellbelove, 2024-03-23:
Working on that right now!

jwellbelove, 2024-04-21:
Fixed 20.38.11

closed 2024-04-21 by jwellbelove