ETLCPP/etl #870: Any plans for `shared_ptr`, `make_unique`, etc?

anthonygclark, 2024-03-25:
Hey John and team,

I've been following ETL and using it for a while; I really appreciate it! I'm curious if there are plans to add more things from STL's `<memory>`? Specifically:

* `shared_ptr` + `make_shared` - seems like there is enough atomic support. Anything else stopping an implementation?
* `make_unique` - curious if there is anything holding ETL back here.

I know smart pointers are opinionated but having them may be nice. I can start implementations and merge requests if you believe there is value.

jwellbelove, 2024-03-25:
Implementing `etl::make_unique` as a drop-in replacement for `std::make_unique` would break the design rules of the ETL by using dynamic memory via `new`. If you need a `make_unique` then it is fairly simple to implement yourself. If you don't want to use heap memory then you will need to overload the `new` operator for the type you want to make a `unique_ptr` to. 

The only other option would be to make an `etl::make_unique` that forces the user to declare _both_ allocator and deleter types in the template parameters.

`etl::shared_ptr` has a related issue in that a `shared_ptr` must contain a pointer to an atomic reference counter that is common to _all_ copies of a `shared_ptr` instance. That is the way that every copy of the `shared_ptr` knows its reference count. This counter, in the STL, would be allocated from heap memory, so every copy of the `shared_ptr` object would point to the same counter. When the reference count reaches zero then both the shared object and reference count would both be deleted. 

As dynamic memory is not used in the ETL then the user would have to manually supply a unique, statically or globally declared, reference counter for each instance of `etl::shared_ptr` that they create. This is not very practical.

On a more general note, the original design aims of the ETL was to create an embedded friendly `enhancement` to the STL, not create a complete clone of it.

RallyTronics, 2024-03-26:
I have my own implementations of shared and unique pointers that use a backing store derived from iallocator_base_t, they only work with a **single type** though.   You have to have a separate store for each different type.

anthonygclark, 2024-03-26:
John,

Thank you so much for the reply. I follow. This is very helpful. I appreciate the work!

@RallyTronics oh interesting. That is very helpful. Any chance you can paste some pseudo code or an example here?

closed 2024-04-15 by jwellbelove