ETLCPP/etl #871: feature request: callback on timer functions to handle changes in the time list

KammutierSpule, 2024-03-27:
would be possible to register a callback on timer functions, so I can be notified when there is a change (insert/remove) on the list?

reason: this way I can wake-up a semaphore, so it will request the time to sleep until next timer... (instead of implement a "fixed polling rate").

jwellbelove, 2024-04-11:
Which particular timer type are you working with? I can send you a possible implementation for you to test.

KammutierSpule, 2024-04-11:
I don't mind to do a code review before testing, do you have it on a branch?

jwellbelove, 2024-04-11:
I've done a quick change to `etl::callback_timer` on `feature/#871-callback-on-timer-functions-to-handle-changes-in-the-time-list`

I added member functions:-
```
void set_event_callbacks(event_callback_type insert, event_callback_type remove)
void clear_event_callbacks()
```
Where `event_callback_type` is `etl::delegate<void(etl::timer::id::type)>`

I haven't written any unit tests for it yet.

jwellbelove, 2024-04-11:
If you want to set just one event callback then set the other to `etl::callback_timer::event_callback{}`

KammutierSpule, 2024-09-03:
I didn't tested but I had a look and seems ok, would be possible to merge?

KammutierSpule, 2025-07-24:
I added a PR https://github.com/ETLCPP/etl/pull/1155 with John suggestions.
I didn't added yet unit tests.
@jwellbelove could you review first, if that is the expected, on all timers, and then I will add unit tests ?

jwellbelove, 2026-09-28:
@KammutierSpule 
Looking at the `tick()` implementations I'm wondering if the callbacks are really optimal.
If a timer is 'repeating' then the callback code is bombarded with a remove_callback, immediately followed by an insert_callback.

What you originally said you wanted was "so it will request the time to sleep until next timer", which may actually be better served in a different way, i.e. by sending the actual time to the next timer timeout in a callback?

jwellbelove, 2026-09-28:
BTW. It doesn't look like I documented these functions anyway.

jwellbelove, 2026-09-29:
This what I had in mind.

[etl-6260c80-Changed the two insert-remove callbacks to one time_to_next callback.patch](https://github.com/user-attachments/files/32794401/etl-6260c80-Changed.the.two.insert-remove.callbacks.to.one.time_to_next.callback.patch)