ETLCPP/etl #875: Side effects concern etl::string

pinwhell, 2024-04-06:
Effects of writing .data() ptr directly? can it expect string to keep working properly? is there any mechanism to simply write the buffer internally directly and not having side effects? also what about that repair() does it has something to do with this concern?

jwellbelove, 2024-04-06:
Like `std::string`, you can modify the contents of the string using the pointer returned from `data()`.
BUT, you must not try to change the length of the string by manipulating the null terminator.

If you have a C function that needs to concatenate characters to an `etl::string`, then you must `resize()` the string to the maximum, let the C function add the characters, and then `resize()` the string to the new length. The first `resize()` will fill the new space with null characters.

`repair()` resets the internal pointer to the buffer after an `etl::string` has been moved by a byte copy such as `memcpy`.

jwellbelove, 2024-04-06:
There is also `trim_to_terminator()` which sets the string size to the distance to the first null terminator.

pinwhell, 2024-04-06:
Cool, this information is very useful, really appreciated...

closed 2024-04-06 by pinwhell

jwellbelove, 2024-04-06:
There is also `data_end()` in `etl::string`, which can make it even simpler.

```
etl::string<11> text = "Hello";

// C function
strcpy(text.data_end(), " World");

text.trim_to_terminator(); // text contains "Hello World"
```

jwellbelove, 2024-04-06:
There is also a `repair()` for `etl::vector` & `etl::deque` to facilitate copying of them by C functions. This can be useful for copying ETL buffers in an RTOS.

pinwhell, 2024-04-06:
yes, very very useful and interesting concept