ETLCPP/etl #876: ETL Functional How to use?

pinwhell, 2024-04-06:
i see, ETL has std::functions, but how to use them?
STL ones allows to "copy construct" a  with a lambda for example

std::function<...> = []{};

what could be the equivalent for ETL?

jwellbelove, 2024-04-06:
The ETL has two types of delegates, `etl::function` & `etl::delegate`.
`etl::function` has no connection with `std::function` and has now been deprecated (it will eventually be removed).

`etl::delegate` is its replacement. `etl::delegate` can handle lambdas to a **limited** extent. The lambda is passed as a reference and its lifetime **must** exist for the lifetime of the delegate. This is because the `etl::delegate` does not allocate any storage for the lambda. `std::function` can copy as it will allocate memory from the heap in which to store it, but as the ETL has a 'no dynamic memory' rule, this cannot be done for `etl::delegate`.

I have an API compatible ETL version of `std::function` on an experimental branch, but it has the same dynamic memory limitations. I have not added it to the main branch, as I don't see that is has any real advantages over `etl::delegate`. 

In fact `etl::delegate` has some advantages over `std::function` in that `etl::delegate` objects are copyable to any other `etl::delegate` object that handles the same function signature. Directly copying between `std::function` objects containing functions with different natures (global vs. member) isn't straightforward.

pinwhell, 2024-04-06:
etl::delegate i tested, 😍, even if there is no connection with std::function, that's i cant ask for more, amazing!!!!!, in the future i as soon as i get free time i will consider start full time contributing to ETL, i love this library

closed 2024-04-06 by pinwhell