ETLCPP/etl #893: Feature Request: Support for String Concatenation Using << Operator

JorgeHSantana, 2024-05-14:
Hello ETLCPP Team,

I am proposing an enhancement to the ETL that would introduce the ability to use the << operator for string concatenation. This feature would simplify string manipulations and align the library closer to C++ STL conventions, providing a more intuitive and familiar interface for string operations.

**Proposed Functions:**

```
etl::istring& operator << (etl::istring& lhs, char rhs) {
    lhs += rhs;
    return lhs;
}

etl::istring& operator << (etl::istring& lhs, const char* rhs) {
    lhs += rhs;
    return lhs;
}

etl::istring& operator << (etl::istring& lhs, etl::istring& rhs) {
    lhs += rhs;
    return lhs;
}

template<typename T>
etl::istring& operator << (etl::istring& lhs, T rhs){
    etl::to_string(rhs, lhs, etl::format_spec().precision(2), true);
    return lhs;
}
```

**Usage Example:**

```
etl::string<100> str;
str << "Hello, " << "World! " << 3.14 << " " << 10;
```

**Extending to Custom Objects:**

```
class myClass {
private:
    int _value {1};
    const char* _name {"myClass"};

public:
    myClass() {}
    friend etl::istring& operator << (etl::istring& lhs, myClass& rhs);
};

etl::istring& operator << (etl::istring& lhs, myClass& rhs) {
    lhs << "myClass: " << rhs._name << " " << rhs._value;
    return lhs;
}
```

With this functionality, users can seamlessly concatenate strings, primitives, and objects into etl::istring instances, enhancing usability and flexibility of the string handling in ETL.

**Collaboration:**
I am willing to contribute to implementing this feature. If acceptable, I would appreciate guidance on how you would like the new code structured, particularly regarding which files should be modified or if new files should be created for this purpose.

Could this be considered for inclusion in ETL? If you're open to it, I can prepare a pull request with these changes.

Thank you for considering this enhancement. I look forward to your feedback and suggestions.

jwellbelove, 2024-05-16:
You seem to be recreating some of the functionality of `etl::string_stream`, but without the statefulness of `etl::string_stream`.

JorgeHSantana, 2024-05-16:
Forgiveness! I hadn't seen that you had this implementation! I'm impressed!

closed 2024-05-16 by JorgeHSantana