ETLCPP/etl #900: Wrong namespace: std instead of etl in algorithm.h:1296

jputcu, 2024-06-04:
Probably a copy paste error:

`etl/algorithm.h:1296:12: std::advance(result, std::distance(middle, last));`

~~~~
In file included from C:/Users/joris/.conan2/p/etl50fc888098f23/p/include/etl/array.h:35:0,
                 from C:/Users/joris/.conan2/p/generbb1ed3e9b7e4a/s/include/general/sha1.h:6,
                 from C:/Users/joris/.conan2/p/generbb1ed3e9b7e4a/s/src/sha1.cpp:1:
C:/Users/joris/.conan2/p/etl50fc888098f23/p/include/etl/algorithm.h: In function 'constexpr typename etl::enable_if<etl::is_forward_iterator<T>::value, TIterator>::type etl::private_algorithm::rotate_general(TIterator, TIterator, TIterator)':
C:/Users/joris/.conan2/p/etl50fc888098f23/p/include/etl/algorithm.h:1296:12: error: 'advance' is not a member of 'std'
       std::advance(result, std::distance(middle, last));
            ^~~~~~~
C:/Users/joris/.conan2/p/etl50fc888098f23/p/include/etl/algorithm.h:1296:12: note: suggested alternative:
In file included from C:/Users/joris/.conan2/p/etl50fc888098f23/p/include/etl/algorithm.h:43:0,
                 from C:/Users/joris/.conan2/p/etl50fc888098f23/p/include/etl/array.h:35,
                 from C:/Users/joris/.conan2/p/generbb1ed3e9b7e4a/s/include/general/sha1.h:6,
                 from C:/Users/joris/.conan2/p/generbb1ed3e9b7e4a/s/src/sha1.cpp:1:
C:/Users/joris/.conan2/p/etl50fc888098f23/p/include/etl/iterator.h:139:24: note:   'etl::advance'
   ETL_CONSTEXPR14 void advance(TIterator& itr, TDistance n)
                        ^~~~~~~
In file included from C:/Users/joris/.conan2/p/etl50fc888098f23/p/include/etl/array.h:35:0,
                 from C:/Users/joris/.conan2/p/generbb1ed3e9b7e4a/s/include/general/sha1.h:6,
                 from C:/Users/joris/.conan2/p/generbb1ed3e9b7e4a/s/src/sha1.cpp:1:
C:/Users/joris/.conan2/p/etl50fc888098f23/p/include/etl/algorithm.h:1296:33: error: 'distance' is not a member of 'std'
       std::advance(result, std::distance(middle, last));
                                 ^~~~~~~~
C:/Users/joris/.conan2/p/etl50fc888098f23/p/include/etl/algorithm.h:1296:33: note: suggested alternative:
In file included from C:/Users/joris/.conan2/p/etl50fc888098f23/p/include/etl/algorithm.h:43:0,
                 from C:/Users/joris/.conan2/p/etl50fc888098f23/p/include/etl/array.h:35,
                 from C:/Users/joris/.conan2/p/generbb1ed3e9b7e4a/s/include/general/sha1.h:6,
                 from C:/Users/joris/.conan2/p/generbb1ed3e9b7e4a/s/src/sha1.cpp:1:
C:/Users/joris/.conan2/p/etl50fc888098f23/p/include/etl/iterator.h:197:77: note:   'etl::distance'
   ETL_CONSTEXPR14 typename etl::iterator_traits<TIterator>::difference_type distance(TIterator first, TIterator last)
                                                                             ^~~~~~~~
~~~~

jwellbelove, 2024-06-04:
Fixed in the latest release

Chiraffollo, 2024-06-04:
Unfortunately, the fix is not included in the latest release. The fix is mentioned in the release notes but the branch of the corresponding pull request seems not to be merged into master branch. In the master branch the issue is still present.

jwellbelove, 2024-06-04:
Oops! I'll fix that later today.

jwellbelove, 2024-06-05:
Fixed in 20.38.17

closed 2024-06-05 by jwellbelove