ETLCPP/etl #904: Feature request: Implement equivalent for std::type_identity<T> in <etl/type_traits.h>

tigran2008, 2024-06-10:
`std::type_identity` was added in C++20 as a way to escape type deduction in templated arguments. https://en.cppreference.com/w/cpp/types/type_identity

Manually implementing it like this seems to work, and although it's just a few lines, it'd be nice if it was built-in.

```cpp
template <typename T>
struct type_identity { typedef T type; };

#if ETL_USING_CPP11
  template <typename T>
  using type_identity_t = typename type_identity<T>::type;
#endif
```

jwellbelove, 2024-06-10:
Do you want to do a Pull Request?

tigran2008, 2024-06-10:
Would I add it to `etl/generators/type_traits_generator.h`?

jwellbelove, 2024-06-10:
Yes, and then run the generator script `generate_type_traits.bat`

tigran2008, 2024-06-10:
Alright, cool! I'm installing cogapp right now that's why it's taking some time. 😅

tigran2008, 2024-06-10:
There we go: https://github.com/ETLCPP/etl/pull/905

closed 2024-06-10 by jwellbelove