ETLCPP/etl #906: `platform.h` error when building with GCC 13 using C++14

jaskij, 2024-06-25:
Working or some minor improvements, and wanted to test them using C++14. Forced it in CMakeLists.txt. This started erroring out in `platform.h` because `char8_t` is a reserved identifier in C++20. The only way to get rid of the error (which seems a false positive given the codebase) was to add

```cmake
add_compile_options(-Wno-c++20-compat)
```

CMake patch to force C++14:

```diff
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -32,6 +32,8 @@ add_library(etl::etl ALIAS ${PROJECT_NAME})
 
 include(GNUInstallDirs)
 
+add_compile_options(-std=c++14)
+
 target_include_directories(${PROJECT_NAME} ${INCLUDE_SPECIFIER} INTERFACE
         $<BUILD_INTERFACE:${CMAKE_CURRENT_SOURCE_DIR}/include>
         $<INSTALL_INTERFACE:${CMAKE_INSTALL_INCLUDEDIR}>
```

Error:

```
[24/303] Building CXX object test/CMakeFiles/etl_tests.dir/murmurhash3.cpp.o
FAILED: test/CMakeFiles/etl_tests.dir/murmurhash3.cpp.o 
/usr/bin/c++ -DETL_DEBUG -I/home/jaskij/projects/deps/etl/test/../include -isystem /home/jaskij/projects/deps/etl/test/UnitTest++/.. -std=gnu++20 -std=c++14 -fno-omit-frame-pointer -fno-common -Wall -Wextra -Werror -Wfloat-equal -Wuseless-cast -Wshadow -Wnull-dereference -MD -MT test/CMakeFiles/etl_tests.dir/murmurhash3.cpp.o -MF test/CMakeFiles/etl_tests.dir/murmurhash3.cpp.o.d -o test/CMakeFiles/etl_tests.dir/murmurhash3.cpp.o -c /home/jaskij/projects/deps/etl/test/murmurhash3.cpp
In file included from /home/jaskij/projects/deps/etl/test/murmurhash3.cpp:10:
/home/jaskij/projects/deps/etl/test/../include/etl/platform.h:361:25: error: identifier ‘char8_t’ is a keyword in C++20 [-Werror=c++20-compat]
  361 |   typedef uint_least8_t char8_t;
```

jaskij, 2024-06-25:
using `ETL_CXX_STANDARD` to set the C++ version causes the same error, with the same workaround

jwellbelove, 2024-06-25:
Try defining `ETL_NO_SMALL_CHAR_SUPPORT` as `0` in etl_profile.h or in the compiler settings.

jaskij, 2024-06-25:
Won't that create issues elsewhere? Since it's simply a forward-compatibility warning.

jaskij, 2024-06-25:
Ah! But it doesn't have `char8_t`.

I'm using GCC 13, but compiling in C++14 mode. This is basically a warning saying that the line *would be* an error if it was compiled in C++20 mode. Which won't happen because of the `#ifdef`s, but GCC seems to miss it.

jwellbelove, 2024-06-27:
We may have to disable this warning, as I can't see any way to solve this other than **not** defining `char8_t` for <C++20, which is not a real solution.

jwellbelove, 2024-06-27:
What is the GCC warning number for this?

PR #909 (2024-07-02): fix c++20-compat false positive
Fix #906
closed 2024-07-03 by jaskij