ETLCPP/etl #911: Speed up builds

jaskij, 2024-07-02:
Something I noticed is that current GHA workflows simply call `make` - which defaults to running the jobs sequentially. Meanwhile, GitHub's public runners [have more than one CPU core](https://docs.github.com/en/actions/using-github-hosted-runners/about-github-hosted-runners/about-github-hosted-runners#standard-github-hosted-runners-for-public-repositories).

Your builds could probably be significantly sped up by changing the `make` call to something like `make -j4`, or more universally `make -j $(nproc)`. It could also be possible to switch over to ninja (which defaults to one job per CPU thread), but I don't think it's installed by default in GHA images.

jaskij, 2024-07-02:
I'll take care of this one once #907 is merged.

It also seems to me that there's a lot of opportunity for deduplication in workflow code, utilizing matrices and environment variables.

jwellbelove, 2024-07-02:
> It also seems to me that there's a lot of opportunity for deduplication in workflow code, utilizing matrices and environment variables.

You're probably correct.
I look forward to seeing what you propose.

PR #912 (2024-08-01): Improve ci
1. Remove duplication in workflow definitions by using matrices. Should make the inline display in PR more readable too.
2. Tell `make` to use all the CPUs. Seems to roughly halve the build time.

Closes #911 

Also: due to the nature of the changes, my branch will have a lot of noise in commits, so I *highly* recommend squash merging.

-----

Subtasks:

- [X] Linux builds - done
- [X] macOS builds - done - needs discussion
- [X] Windows - done
- [ ] caching - left for future
- [X] explore `cmake --build` - won't gain anything with it

----

Before merging:

- [ ] `-Wno-vla-cxx-exte [... cut]