ETLCPP/etl #917: Bug: unaligned_type doesn't compile with floating point types

enbyted, 2024-07-10:
`unaligned_type<T, Entian>` claims to support any `T` that is either integral or floating point. There are also type aliasses for float, double and long double.

Unfortunately, the `unaligned_copy` algorithm does not support those correctly, fails with the following error:

```
etl/include/etl/unaligned_type.h:464:54: error: invalid operands of types ‘float’ and ‘unsigned int’ to binary ‘operator>>’
   464 |           store[1] = static_cast<storage_type>(value >> (1U * CHAR_BIT));
       |                                                ~~~~~~^~~~~~~~~~~~~~~~~~
```

jwellbelove, 2024-08-09:
I'm looking at removing the equality and inequality operators as 
1) They do not add any real functionality to `etl::unaligned_type`.
2) If they are included then the other numerical operators should logically be included.
3) They hide the conversion to and from the underlying type that is going on to enable the comparison. This is especially true of the floating point types as the they can have an implementation defined amount of padding.

jwellbelove, 2024-08-10:
Fixed 20.39.3

closed 2024-08-10 by jwellbelove