ETLCPP/etl #920: Define `ETL_USING_LEGACY_VARIANT` as a convenience macro for code that tries to support both new and legacy variants

tigran2008, 2024-07-15:
Right now one has to repeat the condition in <etl/variant.h>, e.g. `#if !ETL_USING_CPP11 || defined(ETL_USE_LEGACY_VARIANT)` but it would be nicer to do `#if ETL_USING_LEGACY_VARIANT`. Obviously, it would be as simple as `#define ETL_USING_LEGACY_VARIANT !ETL_USING_CPP11 || defined(ETL_USE_LEGACY_VARIANT)`

tigran2008, 2024-07-15:
(regarding supporting both new and old variants, it would be nicer if the new API had some stuff from the older one, e.g. `is_type` for the sake of reducing conditional compilation)

tigran2008, 2024-07-18:
Well, I personally haven't needed another test method myself yet, but if it wouldn't be difficult, I guess having `is_supported_type`, `is_same_type` and `is_valid` might be useful for others. (What does `is_valid` do, though?)

jwellbelove, 2024-07-19:
It's impossible to have an `is_valid` that works the same as the legacy as, like `std::variant`, it default constructs the first type. The only way would be for the variant's first type to be `etl::monostate`, and detect this.

In the legacy variant, a default constructed object is 'invalid'.

jwellbelove, 2024-08-10:
Fixed 20.39.3

closed 2024-08-10 by jwellbelove