ETLCPP/etl #936: HFSM Self Transition

Chippychap, 2024-07-30:
#696 introduced `ifsm_state::Self_Transition` for the FSM implementation, but not for the HFSM implementation.
Expected behavior is to also reactivate (exit, enter) the current state when `ifsm_state::Self_Transition` is returned by `process_event` / `on_event`
Now, this may be a bit more complex as there are also parent state handling to take care of. As what should happen if the parent returns the `ifsm_state::Self_Transition` variable. Should than the parent be reactivated as well?

Also, but this may need to be a different issue but somewhat related. Returning the `STATE_ID` from the parent when it processes an event causes some weird transitions as it partially reactivates the states.
For example, in the motor control example if the `Running` state would accept the `SetSpeed` event and returns the `STATE_ID`. It would then reenter the `Running` state without exiting it first and activate its initial substate `WindingUp`

jwellbelove, 2025-05-27:
If I remember correctly (`hfsm` was a pull-request contribution way back in 2017), `Self_Transition` is not valid for `hfsm`, for the reasons you state.
If think the one to use is in this case is `No_State_Change`.
The weirdness happens because `hfsm` is a extension of `fsm`.

jwellbelove, 2025-05-27:
hotfix/#936-HFSM-Self-Transition

jwellbelove, 2025-05-27:
I've added self transition support the the HFSM.
It will be in the next release.

closed 2025-05-27 by jwellbelove