ETLCPP/etl #937: ETL_USING_LEGACY_VARIANT Is always getting evaluated to 1

drewr95, 2024-07-30:
It seems that ETL_USING_LEGACY_VARIANT is always getting evaluated to 1 for me.  I do not have ETL_USE_LEGACY_VARIANT defined, so the !ETL_USING_CPP11 must be getting evaluated to true.  I believe this is because these macros are defined before determining compiler versions.
 
[image]

drewr95, 2024-07-30:
Looks like VSCode is confirming this:
[image]

If we don't determine compiler version til line 153 (profiles/determine_compiler_version.h), how can ETL_USING_CPP11 be evaluated?

drewr95, 2024-07-30:
Yeah, can confirm, if I move that macro below determining compiler version my builds work again

jwellbelove, 2024-08-10:
Fixed 20.39.3

closed 2024-08-10 by jwellbelove