ETLCPP/etl #941: Add templated ticker type for callback timers

Sedictious, 2024-08-02:
Currently the callback timers support a fixed timer type of `uint32_t` which fixes the resolution. I think it would be useful to add a template parameter to allow for configurable timer size.

I was thinking about adding something like `timer_size` along the lines of what is supported for `etl::queue` with `memory_model` which could default to `uin32_t`, thus not breaking any backwards compatibility. Would that be something of interest or is there any limiting factor I am missing?

jwellbelove, 2024-08-02:
I'll take a look and see if it's viable.

Sedictious, 2024-08-02:
@jwellbelove I am dealing with the change locally so it would be possible for me to clean up and and prepare a PR  following the style of `memory_model `  if it would help

jwellbelove, 2024-08-02:
OK, give it a go. I'll be interested in what you come up with.
Don't forget there are multiple timer classes.