ETLCPP/etl #942: Warning on pragma `c++20-compat`

jputcu, 2024-08-02:
Using the latest `etl-20.39.0`, `avr-gcc-7.3` (official microchip) I get many of the following warnings:
~~~~
C:/Users/joris/.conan2/p/etlb996c63d2f6b8/p/include/etl/private/diagnostic_cxx_20_compat_push.h:38:34: warning: unknown option after '#pragma GCC diagnostic' kind [-Wpragmas]
   #pragma GCC diagnostic ignored "-Wc++20-compat"
                                  ^~~~~~~~~~~~~~~~
~~~~

jputcu, 2024-08-02:
Apparently this option is only known to gcc>=10.

jwellbelove, 2024-08-02:
Can you replace the contents of `etl/private/diagnostic_cxx_20_compat_push.h` with

```
#if !defined(__AVR__)
  #if defined(__GNUC__) && !defined(__clang__) && !defined(__llvm__)
    #pragma GCC diagnostic push 
    #pragma GCC diagnostic ignored "-Wc++20-compat"
  #endif

  #if defined(__clang__) || defined(__llvm__)
    #pragma clang diagnostic push 
    #pragma clang diagnostic ignored "-Wc++20-compat"
  #endif
#endif
```

jputcu, 2024-08-02:
Warnings are gone now, but I don't know if this is the cleanest solution.
Will this be sufficient for older native compilers? I saw a post on Stackoverflow (https://stackoverflow.com/questions/3378560/how-to-disable-gcc-warnings-for-a-few-lines-of-code) which advises the following: "For most compilers it is often a good idea to check the compiler version before trying to disable it"

I did the test and the following also works:
~~~~
#if defined(__GNUC__) && (__GNUC__ >= 10) && !defined(__clang__) && !defined(__llvm__)
  #pragma GCC diagnostic push 
  #pragma GCC diagnostic ignored "-Wc++20-compat"
#endif
~~~~

jwellbelove, 2024-08-02:
Yes, that was just a quick test to see it it worked.
I'm leaving for a long weekend away in a couple of hours, so I won't be able to look at this until next Tuesday. If you cab come up with an improved solution, then you could always create a PR.

jputcu, 2024-08-02:
I'm not in a hurry, enjoy your weekend

jwellbelove, 2024-08-10:
Fixed 20.39.3

closed 2024-08-10 by jwellbelove