ETLCPP/etl #951: can't get call_if to work because of etl::optional

KammutierSpule, 2024-09-04:
while using delegate call_if. it works if I call the delegate directly:

`error: cannot convert 'etl::enable_if_t<true, etl::optional<esp_modem::command_result> >' {aka 'etl::optional<esp_modem::command_result>'} to 'esp_modem::command_result' in return`

should it work?

jwellbelove, 2024-09-07:
What is type `esp_modem::command_result`?

jwellbelove, 2024-09-07:
What type are you using as the result of `call_if`?

There is an implicit conversion from `optional` to `bool`, but not to the contained type.

This code compiles for me.

```
enum class command_result 
{
  OK,
  FAIL,
  TIMEOUT
};

static command_result function_that_returns_command_result() 
{
  return command_result::OK;
}

etl::delegate<command_result()> d1 = etl::delegate<command_result()>::create<function_that_returns_command_result>();

etl::optional<command_result> opt = d1.call_if();

command_result result = opt.value();
```

KammutierSpule, 2024-09-19:
I got another one
```
typedef etl::delegate<bool(const uint8_t* a_data, size_t a_data_size)> MyDelegateType;
```
```
  bool WriteToChannel(const uint8_t* a_rx_data, size_t a_rx_data_size) {
      return mydelegate_.call_if(a_rx_data, a_rx_data_size);
}
```
`error: cannot convert 'etl::enable_if_t<true, etl::optional<bool> >' {aka 'etl::optional<bool>'} to 'bool' in return`

This is a ESP32 IDF project.
This sounds to be some compiler setting or limitation.

VzdornovNA88, 2024-09-20:
> I got another one
> ```
> typedef etl::delegate<bool(const uint8_t* a_data, size_t a_data_size)> MyDelegateType;
> ```
> ```
>   bool WriteToChannel(const uint8_t* a_rx_data, size_t a_rx_data_size) { [...]

this follows the behavior of the standard std::optional type, which does not allow implicit type conversions except in the logical context of the if statement, due to the goals of improving the type safety of code in which there is no place for std::optional casts to integral types. This is provided by the explicit specifier for the conversion operator.

jwellbelove, 2024-09-20:
Are you compiling for C++03?
`operator bool()` is implicit for C++03 and explicit for C++11 and above.
This is due to the fact that conversion operators cannot be explicit for C++03.

KammutierSpule, 2024-09-20:
> Are you compiling for C++03 or >= C++11?
I believe it should be >=C++11 but didn't check.. or I'm not aware

closed 2024-11-05 by jwellbelove