ETLCPP/etl #958: Can call `get_succesor()` method of `fsm` class without setting a successor.

kingtom22, 2024-09-17:
Hi,

Firstly, thank You for developing and maintaining this library, it is very useful.
I found that `get_succesor()` method of `fsm` can be called without setting a successor and the function will dereference a `nullptr` in that case. It is not written in the documentation that the user of the framework must call the `has_succesor()` method before using the `get_succesor()` method. I recommend to throw an exception or add this requirement to the documentation.

jwellbelove, 2024-09-17:
I shall look at adding an ETL_ASSERT

jwellbelove, 2024-09-18:
It may also be viable to make the return an `etl::optional`.

jwellbelove, 2024-09-18:
I've just checked, and `get_successor()` asserts an `etl::successor_invalid` if no successor has been set.

kingtom22, 2024-09-19:
I see, in version [`19.5.0`](https://github.com/ETLCPP/etl/tree/19.5.0) etl::succesor has been added, and that implements that check before using pointer of the successor. The issue can be closed.
Thank You, for Your help!

closed 2024-09-19 by kingtom22