ETLCPP/etl #960: string insert vs assign

KammutierSpule, 2024-09-25:
It looks insert does not "repair" the string. should it? or should put on documentation something about it?

jwellbelove, 2024-09-25:
What do you mean by "repair"?

KammutierSpule, 2024-09-25:
the test I did, I used insert and it didn't change (update) the size. 
Does it expects the repair() to be called?

jwellbelove, 2024-09-25:
`repair()` is called by the user when a string has been moved by a low level C copy like `memcpy`. It just re-attaches the buffer to the internal pointer.

Can you give an example of an insert not updating the size?

jwellbelove, 2024-09-25:
When you call `test.insert(0, buffer, buffer + 5)` you are calling this function
```
template <typename TIterator>
iterator insert(const_iterator position, TIterator first, TIterator last)
```
It is expecting the first parameter to be the iterator position where you want the string to be inserted. You are sending it a null iterator.

I assume you meant to call `test.insert(0, buffer, 5);`

It maybe worthwhile mw looking at trying to disable the insert if any of the iterators are integrals.

closed 2024-10-22 by jwellbelove