ETLCPP/etl #967: Bug: circular reference in callback_timer_* when start() is called inside callback for repeating timers

ronaldd2, 2024-10-07:
The callback_timer_locked, callback_timer_atomic and callback_timer_interrupt is restarting the timer automatically **after** the callback is handled for repeating timers. If inside the callback the period is changed the timer is stopped, so normally you should call a start timer to start it again.  When doing this from within in the callback, the timer is inserted twice in the list and a circular reference is created in the linked list. 

The superseded callback_timer doesn't have this problem as there the timer is inserted before the callback is called. I see this behavior is changed 2 years ago in the other versions. Can you remember what the rationale for this change was?

Maybe changing the period inside the callback for a repeating timer is not intended, but this is not documented.  Also it's not possible to stop a timer in the callback as it will be started automatically after the call back.

The easiest fix, I think, is to skip the automatic insert if the timer.is_active() is set, as normally it not possible to do an insert twice.
Moving the insert back is not a real option I think as this gives totally different behavior for existing code.

Thanks in advance

KammutierSpule, 2024-10-07:
Hi @ronaldd2 would you like to have a look to see if this is related or duplicated of
https://github.com/ETLCPP/etl/issues/954

KammutierSpule, 2024-10-07:
There is also this one:

https://github.com/ETLCPP/etl/issues/952

because "If inside the callback the period is changed the timer is" changed, and you are using a mutex protected timer, then you will get a deadlock.

itavero, 2024-11-05:
> The superseded callback_timer doesn't have this problem as there the timer is inserted before the callback is called. I see this behavior is changed 2 years ago in the other versions. Can you remember what the rationale for this change was?

@jwellbelove Can you perhaps shed a light on this?

jwellbelove, 2024-11-07:
I'll take a look as soon as I can.

itavero, 2026-09-23:
@rolandreichweinbmw Confirmed. I reviewed your fix and it resolves the issue correctly for `callback_timer_atomic`/`callback_timer_locked`/`callback_timer_interrupt`. The `!timer.is_active()` guard correctly prevents the trailing unconditional reinsert in `tick()` from double-inserting a timer that the callback itself already restarted via `start()`, which is exactly what caused the circular reference.

I validated the same underlying scenario (repeating timer calling `start()` on itself from within its own callback) independently in our codebase using `callback_timer_deferred_locked`, which isn't affected because it reinserts the timer *before* dispatching the callback rather than after, so no conflicting reinsert can occur there. That gave me good confidence in the root-cause analysis behind your patch.

Happy to see this merged.