ETLCPP/etl #970: CAPACITY in `basic_string` could be constexpr on the platforms that are supported

brawner, 2024-10-11:
Currently `CAPACITY` in the basic_string class is listed as `const size_type CAPACITY`. For platforms that support constexpr, it wouldn't need to be part of its memory footprint if it were constexpr.

brawner, 2024-10-11:
I was also wondering then if `max_size()` could be a static function?

jwellbelove, 2024-10-22:
`CAPACITY` cannot be `constexpr` as its value is passed from the derived classes `ibasic_string` and `string_base`. These two classes are 'size agnostic' in that they are common code shared between all sizes of string, in `string_base`'s case, all sizes and character types.
This is also the reason why `max_size()` and `capacity()` cannot be static.

`etl::basic_string<char, 10>` -> `etl::ibasic_string<char>` -> `string_base`

The top level, size dependent, string classes expose a static constant `MAX_SIZE`.

I could maybe add `constexpr` `max_size()` and `capacity()` overloads to the top level classes?

brawner, 2024-10-22:
Thanks for the response, I see that such a change would not be possible without breaking changes.

closed 2024-11-05 by jwellbelove