ETLCPP/etl #972: template not allowed warning in parameter pack

jputcu, 2024-10-29:
On `etl-20.39.4` using ZakKemble `avr-gcc-14` I get the following warning:

~~~~
/include/etl/parameter_pack.h:141:56: warning: keyword 'template' not allowed in declarator-id [-Wpedantic]
  141 |   constexpr size_t parameter_pack<TTypes...>::template index_of_type<T>::value;
      |                                                        ^~~~~~~~~~~~~~~~
~~~~

Note that this warning doesn't occur on `avr-gcc-7.3`

jwellbelove, 2024-10-29:
I think I've come across issues like this before where one compiler rejects it, and another requires it.

jwellbelove, 2024-11-11:
Can you try replacing parameter_pack.h with the one in this zip file?

[parameter_pack.zip](https://github.com/user-attachments/files/17700862/parameter_pack.zip)

jwellbelove, 2024-11-11:
Can you check both avr-gcc-7.3 & avr-gcc-14?

jputcu, 2024-11-11:
I just tried on both and the warning is gone now.
Would it be an idea to extend your CI to compile the unittests for AVR using the avr-g++7.3 (used by Arduino and Microchip)? This would catch many of the warnings I got in the past.
One step further could even be to run the unittests using an AVR simulator?

jwellbelove, 2025-02-23:
Fixed 20.40.0

closed 2025-02-23 by jwellbelove