ETLCPP/etl #973: Macro conflict with RESERVED

mike919192, 2024-11-11:
Hi,

I ran into a macro conflict recently between one of my C system libraries and etl.  My C library has `RESERVED` defined and that name is also used these 2 files.  In both cases it looks to be used only internally
- queue_spsc_atomic.h
- bip_buffer_spsc_atomic.h

I have temporarily worked around the issue by 
```
//workaround for macro conflict
#ifdef RESERVED
#undef RESERVED
#endif
#include "etl/queue_spsc_atomic.h"
```

Would it be possible to either change the case of the variable or give it an ETL_ prefix?  I can make a PR if either one is acceptable.

Thanks

jwellbelove, 2024-11-11:
I have a long term plan to change all of the constants to This_Style_Case to remove the chance of clashes with C macros.

mike919192, 2024-11-11:
OK thanks, would you want a PR for the 2 files I listed?  It looks like there are maybe some other internal constants (`RESERVED_SIZE`) that could also be updated.  There is also (`MAX_SIZE`) but that is public and would be breaking change.

mike919192, 2025-02-28:
@jwellbelove I noticed this is fixed in the current release.  Thanks

closed 2025-02-28 by mike919192