ETLCPP/etl #979: QueuedMessageRouter does not work : message are not sent to queue

bizulk, 2024-11-26:
Running under ubuntu 22 linux desktop (amd64), this is the output result : 
```
  Received message 1 : '1'
  Received message 1 : '2'
  Received message 2 : '1.2'
  Received message 2 : '3.4'
  Received message 3 : 'Hello'
  Received message 3 : 'World'
  Received unknown message 4
```
After running in debug mode I found out that we never jump into the overrided receive that use the generic message, but the templated one. The only way to force going to the overrided receive in the example is to static cas the message to the generic type (or declare a generic variable pointing to the created message) : 

```
  etl::send_message(router, static_cast<const etl::imessage&>(m1));
  etl::send_message(router, static_cast<const etl::imessage&>(Message1(2)));
  etl::send_message(router, static_cast<const etl::imessage&>(m2));
  etl::send_message(router, static_cast<const etl::imessage&>(Message2(3.4)));
  etl::send_message(router, static_cast<const etl::imessage&>(m3));
  etl::send_message(router, static_cast<const etl::imessage&>(Message3("World")));
  etl::send_message(router, static_cast<const etl::imessage&>(Message4()));
 
```

Then the output is as expected : 

```
Queueing message 1
Queueing message 1
Queueing message 2
Queueing message 2
Queueing message 3
Queueing message 3
Ignoring message 4

Processing message 1
  Received message 1 : '1'
Processing message 1
  Received message 1 : '2'
Processing message 2
  Received message 2 : '1.2'
Processing message 2
  Received message 2 : '3.4'
Processing message 3
  Received message 3 : 'Hello'
Processing message 3
  Received message 3 : 'World'
```

aw-engineer, 2024-11-29:
I also see this with
- ubuntu 22.04
- gcc 11.4.0
- -std=gnu++14

jwellbelove, 2024-11-30:
The issue is the line the `using Base_t::receive;` in `examples\QueuedMessageRouter\QueuedMessageRouter.cpp`

With it there, the compiler selects the base template member function, instead of the top level virtual function. The template member functions were added a while ago to enable more efficient calls when using concrete message types.

I have removed it from the example code.

bizulk, 2024-11-30:
Hello.
indeed, that works with this instruction removed, .
I was confused by this instruction. Could you explain why was this instructions added in the past ?
thanks.

jwellbelove, 2024-11-30:
It's used to give visibility of a base class function that been overloaded by the derived.
In the original code, before the introduction of the templated member functions in the base class, it was actually unnecessary.
After the new code was added, the base class template functions were made visible by the `using`, and the compiler decided that they were a better fit than the derived class's virtual `receive` function, so it called that one instead.

jwellbelove, 2025-02-23:
Fixed 20.40.0

closed 2025-02-23 by jwellbelove