ETLCPP/etl #981: Unable to use etl::size on all platforms

jputcu, 2024-12-01:
I wanted to replace my `sizeof(array)/sizeof(array[0])` with a standard `std::size(array)`.

ETL provides `etl::size(array)` for my AVR because I don't have an STL, however, the same code should also compile on my PC which uses STL, then I don't get `etl::size(array)`.

Current workaround is using ETL_ARRAY_SIZE.

I think the cleanest solution would be to only provide `std::size(array)` for non-stl targets. I know this is a sensitive subject but if something is implemented exactly the same:
- giving it the same name doesn't need documentation (maybe just a link)
- I don't need macro magic to make my library work without ETL
- no ETL_OR_STD
- every C++ developer will recognize the function.

(I think `etl::move`, `etl::forward`,... would also benefit from this).

jwellbelove, 2024-12-01:
There are a few ETL functions that have this issue, and it's down to 'Argument Dependent Lookup' (ADL) rules.

What happens is that the compiler would see both the STL's and the ETL's variant of the function and complain about an ambiguous function call.

The functions that can suffer from this are `swap`, `begin`, `end`, `size` (there may be one or two others).

Putting these into the `std` name space would solve the problem, but break backwards compatibility, as the ETL versions could not exist at the same time.

jputcu, 2024-12-01:
It would be nice to only include a specific subdirectory containing `std` items for the `non-stl` targets.

jputcu, 2025-02-10:
Because I prefer using `std` to make my libraries truely cross platform without depending on etl I've switched to using https://github.com/modm-io/avr-libstdcpp.
So now I have my `std::size()`.
I'll keep using etl for expensive or special functionality e.g. `etl::vector` and `etl::delegate`

closed 2025-02-10 by jputcu

jwellbelove, 2025-02-10:
Actually, I've just have a thought that maybe a using directive could work? (Untested, there may be issues)

```
#if ETL_NOT_USING_STL
namespace std
{
  template<typename TContainer>
  using etl::size;
}
#endif
```