ETLCPP/etl #984: Update Base64 Documentation

rijesha, 2024-12-02:
Hi John.
Very much appreciate the library. When you have time could you update the instructions for using base64?
https://www.etlcpp.com/base64.html
Thanks

jwellbelove, 2024-12-02:
In what way?
We're you looking for an example of every permutation of parameter types?

rijesha, 2024-12-03:
I'm wasn't sure how to implement base64 encoding because I did not know that there were built in permutations. When I used the latest release I got an error that base64::encode doesn't exist. I went through the code and saw that there is a new ibase64_encoder. I initially presumed I needed to implement an interface to do the base64 encoding. So then I ended up going to release 20.38.16 which still worked with the example code. I think I saw some of the permutations, but was not sure if they can be used in the same way as in the original example.

I think the instructions should mention that these permutations exist and in the example code instead of `etl::base64::encode` it should be `etl::base64_/*permutation*/_encoder::encode` or something like that.

jwellbelove, 2025-02-23:
Fixed 20.40.0

closed 2025-02-23 by jwellbelove