ETLCPP/etl #993: Need "lighter weight" ETL_ASSERT

pwsoftwaredesigns, 2024-12-13:
I am using ETL on an MCU with a decently small flash size (64KB).  I have bee digging through the compiler MAP file to decrease the size as much as possible for my debug build.  I still want to have minimal optimization (achieved by -Og) and still have some form of assert

During my investigation, I cam across a whole chunk of functions that were larger than I expected.  Looking into the list file (annotated assembly output), I found (one of many as an example) the following that uses an ETL data structure (a variant_pool in this case):

```
08001b10 <tw::SmartPool<32u, led::EExtSyncRise, led::EExtSyncFall, led::EExtSyncHeartbeat, led::ESyncStart, led::EMagnetTest, led::EStrobeSync, led::EToneSilence, led::EToneSync, led::EToneReset, led::ETimeout, led::ETemp4, led::ETemp3>::makeUnique<led::ETemp3>()::{lambda(void*, void const*)#1}::_FUN(void*, void const*)>:
 8001b10:	b570      	push	{r4, r5, r6, lr}
 8001b12:	b084      	sub	sp, #16
 8001b14:	0004      	movs	r4, r0
 8001b16:	000d      	movs	r5, r1
class ETemp3: public ::tw::hsm::Event<ETemp3>
 8001b18:	4b38      	ldr	r3, [pc, #224]	; (8001bfc <tw::SmartPool<32u, led::EExtSyncRise, led::EExtSyncFall, led::EExtSyncHeartbeat, led::ESyncStart, led::EMagnetTest, led::EStrobeSync, led::EToneSilence, led::EToneSync, led::EToneReset, led::ETimeout, led::ETemp4, led::ETemp3>::makeUnique<led::ETemp3>()::{lambda(void*, void const*)#1}::_FUN(void*, void const*)+0xec>)
 8001b1a:	610b      	str	r3, [r1, #16]
 8001b1c:	4b38      	ldr	r3, [pc, #224]	; (8001c00 <tw::SmartPool<32u, led::EExtSyncRise, led::EExtSyncFall, led::EExtSyncHeartbeat, led::ESyncStart, led::EMagnetTest, led::EStrobeSync, led::EToneSilence, led::EToneSync, led::EToneReset, led::ETimeout, led::ETemp4, led::ETemp3>::makeUnique<led::ETemp3>()::{lambda(void*, void const*)#1}::_FUN(void*, void const*)+0xf0>)
 8001b1e:	610b      	str	r3, [r1, #16]
 8001b20:	4b38      	ldr	r3, [pc, #224]	; (8001c04 <tw::SmartPool<32u, led::EExtSyncRise, led::EExtSyncFall, led::EExtSyncHeartbeat, led::ESyncStart, led::EMagnetTest, led::EStrobeSync, led::EToneSilence, led::EToneSync, led::EToneReset, led::ETimeout, led::ETemp4, led::ETemp3>::makeUnique<led::ETemp3>()::{lambda(void*, void const*)#1}::_FUN(void*, void const*)+0xf4>)
 8001b22:	610b      	str	r3, [r1, #16]
      intptr_t distance = p - p_buffer;
 8001b24:	6803      	ldr	r3, [r0, #0]
 8001b26:	1ac8      	subs	r0, r1, r3
      bool is_within_range = (distance >= 0) && (distance <= intptr_t((Item_Size * Max_Size) - Item_Size));
 8001b28:	d407      	bmi.n	8001b3a <tw::SmartPool<32u, led::EExtSyncRise, led::EExtSyncFall, led::EExtSyncHeartbeat, led::ESyncStart, led::EMagnetTest, led::EStrobeSync, led::EToneSilence, led::EToneSync, led::EToneReset, led::ETimeout, led::ETemp4, led::ETemp3>::makeUnique<led::ETemp3>()::{lambda(void*, void const*)#1}::_FUN(void*, void const*)+0x2a>
 8001b2a:	6963      	ldr	r3, [r4, #20]
 8001b2c:	3b01      	subs	r3, #1
 8001b2e:	6922      	ldr	r2, [r4, #16]
 8001b30:	4353      	muls	r3, r2
 8001b32:	4298      	cmp	r0, r3
 8001b34:	dd0b      	ble.n	8001b4e <tw::SmartPool<32u, led::EExtSyncRise, led::EExtSyncFall, led::EExtSyncHeartbeat, led::ESyncStart, led::EMagnetTest, led::EStrobeSync, led::EToneSilence, led::EToneSync, led::EToneReset, led::ETimeout, led::ETemp4, led::ETemp3>::makeUnique<led::ETemp3>()::{lambda(void*, void const*)#1}::_FUN(void*, void const*)+0x3e>
 8001b36:	2600      	movs	r6, #0
 8001b38:	e000      	b.n	8001b3c <tw::SmartPool<32u, led::EExtSyncRise, led::EExtSyncFall, led::EExtSyncHeartbeat, led::ESyncStart, led::EMagnetTest, led::EStrobeSync, led::EToneSilence, led::EToneSync, led::EToneReset, led::ETimeout, led::ETemp4, led::ETemp3>::makeUnique<led::ETemp3>()::{lambda(void*, void const*)#1}::_FUN(void*, void const*)+0x2c>
 8001b3a:	2600      	movs	r6, #0
      bool is_valid_address = ((distance % Item_Size) == 0);
 8001b3c:	6921      	ldr	r1, [r4, #16]
 8001b3e:	f7fe fb69 	bl	8000214 <__aeabi_uidivmod>
      return is_within_range && is_valid_address;
 8001b42:	2e00      	cmp	r6, #0
 8001b44:	d005      	beq.n	8001b52 <tw::SmartPool<32u, led::EExtSyncRise, led::EExtSyncFall, led::EExtSyncHeartbeat, led::ESyncStart, led::EMagnetTest, led::EStrobeSync, led::EToneSilence, led::EToneSync, led::EToneReset, led::ETimeout, led::ETemp4, led::ETemp3>::makeUnique<led::ETemp3>()::{lambda(void*, void const*)#1}::_FUN(void*, void const*)+0x42>
 8001b46:	2900      	cmp	r1, #0
 8001b48:	d004      	beq.n	8001b54 <tw::SmartPool<32u, led::EExtSyncRise, led::EExtSyncFall, led::EExtSyncHeartbeat, led::ESyncStart, led::EMagnetTest, led::EStrobeSync, led::EToneSilence, led::EToneSync, led::EToneReset, led::ETimeout, led::ETemp4, led::ETemp3>::makeUnique<led::ETemp3>()::{lambda(void*, void const*)#1}::_FUN(void*, void const*)+0x44>
 8001b4a:	2600      	movs	r6, #0
 8001b4c:	e002      	b.n	8001b54 <tw::SmartPool<32u, led::EExtSyncRise, led::EExtSyncFall, led::EExtSyncHeartbeat, led::ESyncStart, led::EMagnetTest, led::EStrobeSync, led::EToneSilence, led::EToneSync, led::EToneReset, led::ETimeout, led::ETemp4, led::ETemp3>::makeUnique<led::ETemp3>()::{lambda(void*, void const*)#1}::_FUN(void*, void const*)+0x44>
      bool is_within_range = (distance >= 0) && (distance <= intptr_t((Item_Size * Max_Size) - Item_Size));
 8001b4e:	2601      	movs	r6, #1
 8001b50:	e7f4      	b.n	8001b3c <tw::SmartPool<32u, led::EExtSyncRise, led::EExtSyncFall, led::EExtSyncHeartbeat, led::ESyncStart, led::EMagnetTest, led::EStrobeSync, led::EToneSilence, led::EToneSync, led::EToneReset, led::ETimeout, led::ETemp4, led::ETemp3>::makeUnique<led::ETemp3>()::{lambda(void*, void const*)#1}::_FUN(void*, void const*)+0x2c>
      return is_within_range && is_valid_address;
 8001b52:	2600      	movs	r6, #0
      ETL_ASSERT(is_item_in_pool(p_value), ETL_ERROR(pool_object_not_in_pool));
 8001b54:	2e00      	cmp	r6, #0
 8001b56:	d00c      	beq.n	8001b72 <tw::SmartPool<32u, led::EExtSyncRise, led::EExtSyncFall, led::EExtSyncHeartbeat, led::ESyncStart, led::EMagnetTest, led::EStrobeSync, led::EToneSilence, led::EToneSync, led::EToneReset, led::ETimeout, led::ETemp4, led::ETemp3>::makeUnique<led::ETemp3>()::{lambda(void*, void const*)#1}::_FUN(void*, void const*)+0x62>
      if (items_allocated > 0) 
 8001b58:	68a3      	ldr	r3, [r4, #8]
 8001b5a:	2b00      	cmp	r3, #0
 8001b5c:	d02d      	beq.n	8001bba <tw::SmartPool<32u, led::EExtSyncRise, led::EExtSyncFall, led::EExtSyncHeartbeat, led::ESyncStart, led::EMagnetTest, led::EStrobeSync, led::EToneSilence, led::EToneSync, led::EToneReset, led::ETimeout, led::ETemp4, led::ETemp3>::makeUnique<led::ETemp3>()::{lambda(void*, void const*)#1}::_FUN(void*, void const*)+0xaa>
        if (p_next != ETL_NULLPTR)
 8001b5e:	6863      	ldr	r3, [r4, #4]
 8001b60:	2b00      	cmp	r3, #0
 8001b62:	d027      	beq.n	8001bb4 <tw::SmartPool<32u, led::EExtSyncRise, led::EExtSyncFall, led::EExtSyncHeartbeat, led::ESyncStart, led::EMagnetTest, led::EStrobeSync, led::EToneSilence, led::EToneSync, led::EToneReset, led::ETimeout, led::ETemp4, led::ETemp3>::makeUnique<led::ETemp3>()::{lambda(void*, void const*)#1}::_FUN(void*, void const*)+0xa4>
          *(uintptr_t*)p_value = reinterpret_cast<uintptr_t>(p_next);
 8001b64:	602b      	str	r3, [r5, #0]
        p_next = p_value;
 8001b66:	6065      	str	r5, [r4, #4]
        --items_allocated;
 8001b68:	68a3      	ldr	r3, [r4, #8]
 8001b6a:	3b01      	subs	r3, #1
 8001b6c:	60a3      	str	r3, [r4, #8]
 8001b6e:	b004      	add	sp, #16
 8001b70:	bd70      	pop	{r4, r5, r6, pc}
 8001b72:	4b25      	ldr	r3, [pc, #148]	; (8001c08 <tw::SmartPool<32u, led::EExtSyncRise, led::EExtSyncFall, led::EExtSyncHeartbeat, led::ESyncStart, led::EMagnetTest, led::EStrobeSync, led::EToneSilence, led::EToneSync, led::EToneReset, led::ETimeout, led::ETemp4, led::ETemp3>::makeUnique<led::ETemp3>()::{lambda(void*, void const*)#1}::_FUN(void*, void const*)+0xf8>)
 8001b74:	9302      	str	r3, [sp, #8]
 8001b76:	2390      	movs	r3, #144	; 0x90
[... 68 lines cut ...]
 8001bf4:	4805      	ldr	r0, [pc, #20]	; (8001c0c <tw::SmartPool<32u, led::EExtSyncRise, led::EExtSyncFall, led::EExtSyncHeartbeat, led::ESyncStart, led::EMagnetTest, led::EStrobeSync, led::EToneSilence, led::EToneSync, led::EToneReset, led::ETimeout, led::ETemp4, led::ETemp3>::makeUnique<led::ETemp3>()::{lambda(void*, void const*)#1}::_FUN(void*, void const*)+0xfc>)
 8001bf6:	f008 fff5 	bl	800abe4 <__cxa_guard_release>
[... cut ...] dard assert macro, I want to use `ETL_LOG_ERRORS`.

However, if looks like the implementation of error logging is what generates all this extra code.

**So, my question is, is there a way (or can there by a way) to implement a lightweight assert for ETL?**

For example, in most of my code, I defined my own assert macro like so:

`#define MY_ASSERT(x) if (!(x)) { assert_function(); }`

You don't get much debugging information, but I tend to use the debugger to back trace where an assert occurred.

jwellbelove, 2024-12-13:
The `invocation` element is static and there should be only one in existence for the whole application.

It is accessed within the error handler via a static member function.

```
static invocation_element& get_invocation_element()
{
  static invocation_element invocation;

  return invocation;
}
```

The invocation code in `etl::error_handler` is a much cut-down version of what goes on inside of `etl::delegate`.

jwellbelove, 2024-12-14:
_For example, in most of my code, I defined my own assert macro like so:
#define MY_ASSERT(x) if (!(x)) { assert_function(); }_

We would need to define a standard 'assert_function' name for this to work with the syntax of `ETL_ASSERT`.

jwellbelove, 2024-12-14:
Or require a macro to be defined by the user?

`#define ETL_ASSERT_FUNCTION my_assert_function`

Then have an option for the macros in `error_handler.h` that defines this.

`#define ETL_ASSERT(b, e) {if(!(b)) {ETL_ASSERT_FUNCTION((e));}}`
etc.

jwellbelove, 2024-12-20:
I've just remembered that the error handler (which contains `static invocation_element invocation;`) is not defined unless `ETL_LOG_ERRORS` is defined.

pwsoftwaredesigns, 2024-12-20:
> Or require a macro to be defined by the user?
> 
> `#define ETL_ASSERT_FUNCTION(e) my_assert_function((e))`
> 
> Then have an option for the macros in `error_handler.h` that defines this.
> 
> `#define ETL_ASSERT(b, e) {if(!(b)) {ETL_ASSERT_FUNCTION((e));}}` etc.

Perhaps it is simple as wrapping the definitions of `ETL_ASSERT` into a `ifndef` guard so that the user can optionally redefine the whole macro.

pwsoftwaredesigns, 2024-12-20:
> I've just remembered that the error handler (which contains `static invocation_element invocation;`) is not defined unless `ETL_LOG_ERRORS` is defined.

For my "debug" builds I have been defining `ETL_LOG_ERRORS` and for my "release" builds I have been defining `ETL_NO_CHECKS`.

The problem comes when using `ETL_LOG_ERRORS` which enables `ETL_ASSERT`.  Based upon your description of how `invocation_element` is implemented, I would expect places where `ETL_ASSERT` is used to be compiled to only a few instructions (e.g., a conditional branch and a function call), however, as you can see from the assembly I posted, the compiler seems to be inlining quite a bit of the invocation code.

Normally I wouldn't have a problem with this, except that all this code is inlined every time I call an ETL method that uses `ETL_ASSERT` (in this case it's when calling `etl::variant_pool::destroy`).

I think what I will do as an experiment is temporarily overwrite `ETL_ASSERT` as I have suggested to see if that truly has an effect on the overall code size.

pwsoftwaredesigns, 2024-12-20:
This is slightly off topic but still related.

It looks as if (in my example) the compiler chooses to inline `ipool::release_item` (and `ipool::is_item_in_pool`).  I believe it chooses to do this because all the methods in `ipool` are defined inline in the class declaration (I have seen this often in my own code).  Since this function is called quite often and it is not a trivial function, it gets inlined quite often, thus leading to bloated code size.  But, it's not templated so there's no reason there can't be a single definition and a branch instead of inlining.

pwsoftwaredesigns, 2024-12-20:
For my tests, I added the following lines to the bottom of `error_handler.h`:

```
#undef ETL_ASSERT
#undef ETL_ASSERT_FAIL

#include <tw/assert.h>
#define ETL_ASSERT(b, e) TW_ASSERT(b)
#define ETL_ASSERT_FAIL(e) TW_ASSERT(false)
```

`TW_ASSERT` is the suggested "simple" assert macro I alluded to that simple is an if statement that calls a static/free function.

After compiling, here is the assembly for the same function I showed in my original question:

```
08001854 <tw::SmartPool<32u, led::EExtSyncRise, led::EExtSyncFall, led::EExtSyncHeartbeat, led::ESyncStart, led::EMagnetTest, led::EStrobeSync, led::EToneSilence, led::EToneSync, led::EToneReset, led::ETimeout, led::ETemp4, led::ETemp3>::makeUnique<led::ETemp3>()::{lambda(void*, void const*)#1}::_FUN(void*, void const*)>:
 8001854:	b570      	push	{r4, r5, r6, lr}
 8001856:	0004      	movs	r4, r0
 8001858:	000d      	movs	r5, r1
class ETemp3: public ::tw::hsm::Event<ETemp3>
 800185a:	4b1a      	ldr	r3, [pc, #104]	; (80018c4 <tw::SmartPool<32u, led::EExtSyncRise, led::EExtSyncFall, led::EExtSyncHeartbeat, led::ESyncStart, led::EMagnetTest, led::EStrobeSync, led::EToneSilence, led::EToneSync, led::EToneReset, led::ETimeout, led::ETemp4, led::ETemp3>::makeUnique<led::ETemp3>()::{lambda(void*, void const*)#1}::_FUN(void*, void const*)+0x70>)
 800185c:	610b      	str	r3, [r1, #16]
 800185e:	4b1a      	ldr	r3, [pc, #104]	; (80018c8 <tw::SmartPool<32u, led::EExtSyncRise, led::EExtSyncFall, led::EExtSyncHeartbeat, led::ESyncStart, led::EMagnetTest, led::EStrobeSync, led::EToneSilence, led::EToneSync, led::EToneReset, led::ETimeout, led::ETemp4, led::ETemp3>::makeUnique<led::ETemp3>()::{lambda(void*, void const*)#1}::_FUN(void*, void const*)+0x74>)
 8001860:	610b      	str	r3, [r1, #16]
 8001862:	4b1a      	ldr	r3, [pc, #104]	; (80018cc <tw::SmartPool<32u, led::EExtSyncRise, led::EExtSyncFall, led::EExtSyncHeartbeat, led::ESyncStart, led::EMagnetTest, led::EStrobeSync, led::EToneSilence, led::EToneSync, led::EToneReset, led::ETimeout, led::ETemp4, led::ETemp3>::makeUnique<led::ETemp3>()::{lambda(void*, void const*)#1}::_FUN(void*, void const*)+0x78>)
 8001864:	610b      	str	r3, [r1, #16]
      intptr_t distance = p - p_buffer;
 8001866:	6803      	ldr	r3, [r0, #0]
 8001868:	1ac8      	subs	r0, r1, r3
      bool is_within_range = (distance >= 0) && (distance <= intptr_t((Item_Size * Max_Size) - Item_Size));
 800186a:	d407      	bmi.n	800187c <tw::SmartPool<32u, led::EExtSyncRise, led::EExtSyncFall, led::EExtSyncHeartbeat, led::ESyncStart, led::EMagnetTest, led::EStrobeSync, led::EToneSilence, led::EToneSync, led::EToneReset, led::ETimeout, led::ETemp4, led::ETemp3>::makeUnique<led::ETemp3>()::{lambda(void*, void const*)#1}::_FUN(void*, void const*)+0x28>
 800186c:	6963      	ldr	r3, [r4, #20]
 800186e:	3b01      	subs	r3, #1
 8001870:	6922      	ldr	r2, [r4, #16]
 8001872:	4353      	muls	r3, r2
 8001874:	4298      	cmp	r0, r3
 8001876:	dd0b      	ble.n	8001890 <tw::SmartPool<32u, led::EExtSyncRise, led::EExtSyncFall, led::EExtSyncHeartbeat, led::ESyncStart, led::EMagnetTest, led::EStrobeSync, led::EToneSilence, led::EToneSync, led::EToneReset, led::ETimeout, led::ETemp4, led::ETemp3>::makeUnique<led::ETemp3>()::{lambda(void*, void const*)#1}::_FUN(void*, void const*)+0x3c>
 8001878:	2600      	movs	r6, #0
 800187a:	e000      	b.n	800187e <tw::SmartPool<32u, led::EExtSyncRise, led::EExtSyncFall, led::EExtSyncHeartbeat, led::ESyncStart, led::EMagnetTest, led::EStrobeSync, led::EToneSilence, led::EToneSync, led::EToneReset, led::ETimeout, led::ETemp4, led::ETemp3>::makeUnique<led::ETemp3>()::{lambda(void*, void const*)#1}::_FUN(void*, void const*)+0x2a>
 800187c:	2600      	movs	r6, #0
      bool is_valid_address = ((distance % Item_Size) == 0);
 800187e:	6921      	ldr	r1, [r4, #16]
 8001880:	f7fe fcc8 	bl	8000214 <__aeabi_uidivmod>
      return is_within_range && is_valid_address;
 8001884:	2e00      	cmp	r6, #0
 8001886:	d005      	beq.n	8001894 <tw::SmartPool<32u, led::EExtSyncRise, led::EExtSyncFall, led::EExtSyncHeartbeat, led::ESyncStart, led::EMagnetTest, led::EStrobeSync, led::EToneSilence, led::EToneSync, led::EToneReset, led::ETimeout, led::ETemp4, led::ETemp3>::makeUnique<led::ETemp3>()::{lambda(void*, void const*)#1}::_FUN(void*, void const*)+0x40>
 8001888:	2900      	cmp	r1, #0
 800188a:	d004      	beq.n	8001896 <tw::SmartPool<32u, led::EExtSyncRise, led::EExtSyncFall, led::EExtSyncHeartbeat, led::ESyncStart, led::EMagnetTest, led::EStrobeSync, led::EToneSilence, led::EToneSync, led::EToneReset, led::ETimeout, led::ETemp4, led::ETemp3>::makeUnique<led::ETemp3>()::{lambda(void*, void const*)#1}::_FUN(void*, void const*)+0x42>
 800188c:	2600      	movs	r6, #0
 800188e:	e002      	b.n	8001896 <tw::SmartPool<32u, led::EExtSyncRise, led::EExtSyncFall, led::EExtSyncHeartbeat, led::ESyncStart, led::EMagnetTest, led::EStrobeSync, led::EToneSilence, led::EToneSync, led::EToneReset, led::ETimeout, led::ETemp4, led::ETemp3>::makeUnique<led::ETemp3>()::{lambda(void*, void const*)#1}::_FUN(void*, void const*)+0x42>
      bool is_within_range = (distance >= 0) && (distance <= intptr_t((Item_Size * Max_Size) - Item_Size));
 8001890:	2601      	movs	r6, #1
 8001892:	e7f4      	b.n	800187e <tw::SmartPool<32u, led::EExtSyncRise, led::EExtSyncFall, led::EExtSyncHeartbeat, led::ESyncStart, led::EMagnetTest, led::EStrobeSync, led::EToneSilence, led::EToneSync, led::EToneReset, led::ETimeout, led::ETemp4, led::ETemp3>::makeUnique<led::ETemp3>()::{lambda(void* [... cut ...]
 80018cc:	0800a2c4 	.word	0x0800a2c4
```

The quantitative improvement for me is the final binary size.  Using `ETL_ASSERT` as is (without exceptions but with `ETL_LOG_ERRORS` defined), my binary compiled to 49232 bytes.  With `ETL_ASSERT` and `ETL_ASSERT_FAIL` defined as above, the binary size dropped to 44044 bytes (a decrease of ~5KB).  On a chip with only 64KB of flash, this is pretty significant (and only a problem on debug builds that have all the extra error checking added in without heavy compiler optimizations enabled).

jwellbelove, 2024-12-20:
> Perhaps it is simple as wrapping the definitions of `ETL_ASSERT` into a `ifndef` guard so that the user can optionally redefine the whole macro.

There are actually six macros to redefine, which can make it a little error prone.

In my code, I added this to the macro definitions:-

```
#elif defined(ETL_USE_ASSERT_FUNCTION)
  #define ETL_ASSERT(b, e) {if(!(b)) {ETL_ASSERT_FUNCTION((e));}}                                 // If the condition fails, calls the assert handler
  #define ETL_ASSERT_OR_RETURN(b, e) {if(!(b)) {ETL_ASSERT_FUNCTION((e)); return;}}               // If the condition fails, calls the assert handler and return
  #define ETL_ASSERT_OR_RETURN_VALUE(b, e, v) {if(!(b)) {ETL_ASSERT_FUNCTION((e)); return (v);}}  // If the condition fails, calls the assert handler and return a value

  #define ETL_ASSERT_FAIL(e) {ETL_ASSERT_FUNCTION((e));}                                          // Calls the assert handler
  #define ETL_ASSERT_FAIL_AND_RETURN(e) {ETL_ASSERT_FUNCTION((e)); return;}                       // Calls the assert handler and return
  #define ETL_ASSERT_FAIL_AND_RETURN_VALUE(e, v) {ETL_ASSERT_FUNCTION((e)); return (v);}          // Calls the assert handler and return a value
```

pwsoftwaredesigns, 2024-12-20:
If `ETL_USE_ASSERT_FUNCTION` is added, would the `ETL_ASSERT_FUNCTION` then be defined in a user-provided `etl_profile.h`?

jwellbelove, 2025-02-23:
Fixed 20.40.0

closed 2025-02-23 by jwellbelove