ETLCPP/etl #995: etl::intrusive_forward_list: push_back() and pop_back()

rolandreichweinbmw, 2024-12-15:
Hi,

at https://www.etlcpp.com/intrusive_forward_list.html , a pop_back() method is documented although it is currently not implemented in master or development.

Therefore, I wonder if an implementation of push_back() and pop_back() for etl::intrusive_forward_list is either intentionally left out, or if they are missing (both probably O(n))?

rolandreichweinbmw, 2024-12-16:
See also #996 .

jwellbelove, 2024-12-16:
Although it's possible to add `push_back()` and `pop_back()` to `etl::intrusive_forward_list` (and `etl::forward_list`), they are not included as their implementations are inefficient, due to the need to step through the list from the start. This is why they are not defined for `std::forward_list`.

In this case it is the documentation that is wrong.

rolandreichweinbmw, 2024-12-16:
Fine for me. Thanks for the clarification. You can just close the PR in this case.

jwellbelove, 2024-12-16:
It's the same reason that the index operator `[]` is not defined for lists; possible, but inefficient.