ETLCPP/etl #997: Request: operator[ ] for containers can be configured to emit an exception

abailey-isi, 2024-12-18:
It would be wonderful if there was a way to configure ETL to emit an exception if accessing an out-of-range element of a container when using the operator [ ] instead of just the at( ) method.

e.g.
[image]

jwellbelove, 2024-12-19:
I could maybe add an optional configuration macro similar to `ETL_CHECK_PUSH_POP`, such as 'ETL_CHECK_INDEX_OPERATOR`

jwellbelove, 2025-02-23:
Fixed 20.40.0

rolandreichweinbmw, 2025-04-30:
Was there an actual branch feature/https://github.com/ETLCPP/etl/issues/997-index-operator-for-containers-can-be-configured-to-emit-an-exception as mentioned in the git log? Or do you prefer a new implementation of this? I think this is a useful feature.

rolandreichweinbmw, 2025-05-01:
Do you prefer to implement this on your side?

I already have a similar implementation here, although on a larger scope: out of bounds check where STL/std containers are defined to not check bounds, including vector access at operator[] and erase(). Therefore, I would rather call the (optional) config macro `ETL_CHECK_BOUNDS`.

I'd like to contribute this via a PR. What do you think?

jwellbelove, 2025-09-21:
This is being addressed in #1175