ETLCPP/etl #1000: Make ETL_TYPEDEF Trivially Copyable

drewr95, 2024-12-24:
I'm getting a clang-tidy warning when passing around an ETL_TYPEDEF(degrees, float) by copy to pass it as const reference.  Doing some research, clang-tidy weighs the check with whether the type is trivially copyable.  I think the way the strong types are generated, there is no need to define the copy ctor or assignment operator.  I believe they are defined in a way that would still require the underlying type to be trivially copyable.

drewr95, 2024-12-24:
I see BOOST's strong typedefs are very similar and also not trivially copyable.  Is there a reason why?  With the way it's implemented I'm not sure why it would need to be defined?  Would it be for if the underlying type wasn't trivially copyable?

jwellbelove, 2024-12-26:
I've tried a few experiments, with both trivially and non-trivially copyable types, with the default constructor, copy constructor, and assignment operator removed or defaulted, and everything appears to work as expected.

ETL_TYPEDEF(uint32_t, type1_t);
ETL_TYPEDEF(std::vector<uint32_t>, type2_t);

bool b1 = std::is_trivially_copyable_v<type1_t>; // true
bool b2 = std::is_trivially_copyable_v<type2_t>; // false

drewr95, 2024-12-27:
> I've tried a few experiments, with both trivially and non-trivially copyable types, with the default constructor, copy constructor, and assignment operator removed or defaulted, and everything [...]

That is awesome news!  I'm hoping to have something in within a couple days if you don't beat me to it :)

jwellbelove, 2025-02-23:
Fixed 20.40.0

closed 2025-02-23 by jwellbelove